rusefi/rusefi #4019: toothed previous time

rusefillc, 2022-03-21:
@eldominio reports a FATAL error

[image]

mck1117, 2022-03-22:
I have a hunch about what this is. We queue trigger input events in fast interrupts, then hand off to another interrupt to process them. That second interrupt processes them in pin number order, not chronological order, so it may be possible to get out of order events.

rusefillc, 2022-03-22:
if that's the case would that mean that ICU is totally broken?

mck1117, 2022-03-22:
No, these boards don't use ICU. This is probably a bug in `digital_input_exti.cpp`

rusefillc, 2022-09-01:
@mck1117 this sounds important?

mck1117, 2022-09-01:
we haven't seen it in a while.....so maybe fixed?

closed 2022-09-01 by rusefillc