rusefi/rusefi #4022: Fahrenheit support

BeerMoneyMotorsports, 2022-03-23:
Users are frequently asking to change the units in TS from C to F.  Please advise if we can have this ability added.

rusefillc, 2022-03-23:
Yes possible we had it previously

But we would need to teach you how to make it happen in .input file

mck1117, 2022-03-23:
It's more than just the .input - you have to change the format in the layout section to use different scale/offset factors too. It's a big hassle to do with the way we generate our configuration.

rusefillc, 2022-03-24:
we have a relevant hack/feature

```
#if LAMBDA
uint8_t[FUEL_RPM_COUNT x FUEL_LOAD_COUNT] autoscale lambdaTable;;"lambda", {1/@@PACK_MULT_LAMBDA_CFG@@}, 0, 0.6, 1.5, 2
#else
uint8_t[FUEL_RPM_COUNT x FUEL_LOAD_COUNT] lambdaTable;;"afr", {1/@@PACK_MULT_AFR_CFG@@}, 0, 0, 25, 1
#endif
```