rusefi/rusefi #4031: Trigger setup in TS is highly confusing: hide operation mode from users?

rusefillc, 2022-04-02:
We already only use "operation mode" for skipped wheel triggers

Question: do we even need it for skipped wheel triggers? shall we just declare that all skipped wheel triggers are always crank-based?

rusefillc, 2022-04-02:
@mck1117 @OrchardPerformance @ElDominio?

ElDominio, 2022-04-03:
No, I do a lot of Toyota skipped tooth on cam by grinding a tooth off the 24 tooth trtigger, it's usually the best choice even thought it has a +1. 

Also, the Aprilia 500cc engine is 48-2 on cam from the factory.

rusefillc, 2022-04-03:
What users see now is called
``
engineConfiguration->ambiguousOperationMode
``

which is passed into ``initializeTriggerWaveform`` method - note how limited usages are already

ElDominio, 2022-04-03:
Operation mode means where the trigger is located? Can't they have an option like say, 

Primary trigger location: Crank/Cam 

Secondary is always cam so that's that, but the primary trigger location should be selctable

commit d3c6c586 (2022-04-03): Trigger setup in TS is highly confusing: hide operation mode from users? fix #4031
closed 2022-04-03 by rusefillc