rusefi/rusefi #4038: Hellen NB2 default tune

rusefillc, 2022-04-03:
@BeerMoneyMotorsports let's use this ticket and see if we can come up with a good process

Ask1) please link 'more desired' tune 
Ask2) please upload current default tune (i.e. take unit, click reset to NB2 and upload as is
Ask3) please link a diff page between those two tunes

rusefillc, 2022-04-10:
more desired tune https://rusefi.com/online/view.php?msq=921
current default tune https://rusefi.com/online/view.php?msq=944
difference report default left desired right

https://rusefi.com/online/diff.php?msq1=944&msq2=921

BeerMoneyMotorsports, 2022-04-11:
You have beaten me to the punch.  yes, there are significant differences between tunes.  The tune you linked would need triggers changed for stock as this car has 36-1 on it

rusefillc, 2022-05-02:
fresh more desired tune: https://rusefi.com/online/view.php?msq=985