rusefi/rusefi #4044: Log4j errors

chuckwagoncomputing, 2022-04-05:
Does anyone know what's up with the Log4j errors in the output of enum2string and ConfigDefinition?
Example:
```
E 220404 215549.768 [main] config - com.devexperts.logging.Log4j2Logging link
java.lang.IllegalStateException: java.lang.ClassNotFoundException: com.devexperts.logging.Log4j2Logging
	at com.devexperts.logging.Logging.configure(Logging.java:193)
	at com.devexperts.logging.Logging.<clinit>(Logging.java:49)
	at com.rusefi.EnumsReader.<clinit>(EnumsReader.java:15)
	at com.rusefi.ToJavaEnum.main(ToJavaEnum.java:19)
Caused by: java.lang.ClassNotFoundException: com.devexperts.logging.Log4j2Logging
	at java.net.URLClassLoader.findClass(URLClassLoader.java:387)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:419)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:352)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:352)
	at java.lang.Class.forName0(Native Method)
	at java.lang.Class.forName(Class.java:264)
	at com.devexperts.logging.Logging.configure(Logging.java:187)
	... 3 more
```

chuckwagoncomputing, 2022-04-05:
This is not fixed
https://github.com/rusefi/rusefi/runs/5826967957?check_suite_focus=true#step:5:11

rusefillc, 2022-04-05:
oh right, we keep compiled .jars - each affected tool would need to be recompiled

commit 0cf28cf9 (2022-04-05): Log4j errors fix #4044
commit 4319b3e6 (2022-04-05): Log4j errors fix #4044
commit 278154ad (2022-04-05): Log4j errors fix #4044
closed 2022-04-05 by rusefillc