rusefi/rusefi #4050: valvetronic pigtail

rusefillc, 2022-04-06:
valvetronic pigtail would be nice

it's located under intake on any E65 750 v8 probably many other similar v8 and even some inline-6

pictured is vertical connector and sensor placed vertically inside engine head, with 3rd bolt accessible only with front engine cover removed

[image]

rusefillc, 2022-04-06:
@duhafnusa4 please assign to yourself since you are a secret member

duhafnusa4, 2022-05-20:
I have no idea how to change from secret to not. Can you please help me figure this out?

duhafnusa4, 2022-05-20:
lol i figured it out. I am now public.! @rusefillc assign me to everything.

duhafnusa4, 2022-06-06:
I have acquired 2 pigtails I think from N52 engines? I will be shipping today.