rusefi/rusefi #4054: Air Conditioning Improvements

ElDominio, 2022-04-08:


rusefillc, 2022-04-09:
https://github.com/rusefi/rusefi/wiki/Lua-Scripting#setacdisabledvalue
https://github.com/rusefi/rusefi/wiki/Lua-Scripting#gettimesinceactogglems

ElDominio, 2024-02-23:
Time cut of A/C:
`cycleTime = Timer.new()
cycleTime : reset()
acCycleCount = 0
acReq = 0

timeSinceAc = getTimeSinceAcToggleMs() / 1000

	

	if (timeSinceAc < 0.5) then
		cycleTime : reset()
	end
	acCycle = cycleTime : getElapsedSeconds()

	if (acState == 1) and (acCycle > 15) then

		setAcDisabled(1)
		if acReq == 1 then
			setIdleAdd(-8)
		else
			setIdleAdd(0)
		end
		cycleTime : reset()
	end
	if (acState == 0) and (acCycle > 7) then
		setAcDisabled(nil)

		setIdleAdd(0)
		cycleTime : reset()
	end
	if acReq == 0 then
		setIdleAdd(0)
	end`
	
I cannot find the pressure control method, I willl rewrite it when I get another h154 car. I monitored AC pressure and setAcDisabled(1) if over/under a certain volltage on the sensor, otherwise i would setAcDisabled(nil)

ElDominio, 2025-01-18:
Almost done, only missing cycle times

closed 2025-01-18 by ElDominio