rusefi/rusefi #4057: ITB tuning support

BeerMoneyMotorsports, 2022-04-09:
I have customer request to add some ITB specific support.  

Specifically ability to have a switching point between AlphaN and Speed Density.  

There has been conversation on the best ways to support ITB. I do not know what is best.  @mck1117 @blundar can guide us in the right direction.  

What more info can i provide to support the request?

kjngineering, 2022-04-09:
The most sensible approach here is to allow >1 fuel methods and corresponding maps.
Then support a 'blending table' between the two.

_e.g. 
Fuel Method 1 = Alpha-N with corresponding Fuelling Map
Fuel Method 2 = Speed-Density with corresponding Fuelling Map

Blend Table 1 = Blend from fuel method 1 to fuel method 2

Blend Table 1 would have axis of load (or TPS) and RPM and the blending would be something like
Fuel Total = (blending value x method 1) + ((1-blending value) x Method 2)_

At the required point you can then switch over to closed-loop fuel based on target lambda value.

The same thing could be duplicated for spark.

This would allow also on the fly blending from gasoline to ethanol fuels (Method 1 = 98, Method 2 = E100) and your blending table would be a switch to select your ethanol percentage - or this could come directly from a flex fuel sensor.

see also: https://github.com/rusefi/rusefi/issues/3114

mck1117, 2022-04-10:
@BeerMoneyMotorsports does this car already run? Have we proven that speed density with our map sampling actually doesn't work? I'd love to see a log of it working/not working.

BeerMoneyMotorsports, 2022-04-10:
Currently it runs but without itb. He can try itb with speed  density but i will ask you to help me with it. If itb doesn’t work, priority needs to be given to making a mixture or AN work.

rusefillc, 2022-04-10:
@BeerMoneyMotorsports how do we make it a two way street?

can we know more about car in advance? can we get clear date for ITB install, etc?

BeerMoneyMotorsports, 2022-04-10:
I can work with knightnight to see about his timeline and itb details.

BeerMoneyMotorsports, 2022-04-11:
Timeframe- Approximately 2 months initial goal.  Ultimately doing whatever possible to make sure it is ready before MATG

NB2 that currently runs and drives on NB2 rev d making around 150whp.  

New setup would include cbr 1000 throttles, vacuum block with hoses to each throttle, Remotely mounted NB IAC.

shadowm60, 2022-04-11:
Last year i used sd with itb, car would run nicely till 2500 rpm, then map is unusable, so back to alpha n.

Alpha n sux on the road since if a bit of blip in map like going uphill you have to push pedal.

Logs and tune are available on reo
https://rusefi.com/online/?engineCode=C20XE&sorts[uploadedMsq]=-1&sorts[uploadedLog]=-1 logs from md_sanci and tune. Right now the car is beeing rebuild and upgraded to mre, so probably in 1mo we will start tuning it again.

JRDMcLAREN, 2022-04-13:
I have ITBs on my car, and It works well wit rusefi.
Fuel strategy is Speed Density, but VE map has TPS axis (override VE table to TPS).
Ignition table use MAP sensor.
Vacuum hoses are splitted and connected to MAP.
https://rusefi.com/online/index.php?vehicleName=Daewoo+Nexia&user_id=923