rusefi/rusefi #4059: MRE QC build broken

rusefillc, 2022-04-09:
```
2022-04-09_13_08_47_605: EngineState: CRITICAL error: aut2 analog input validation failed 9.274291992 vs 6.199999809 20220315@microRusEFI
2022-04-09_13_08_47_713: EngineState: Sensor "AuxTemp1": Raw value: 2.73 Valid: Yes Converted value 13.80
2022-04-09_13_08_47_713: EngineState:     2.73 volts -> 3249.0 ohms with supply voltage 5.00 and pullup 2700.0.
2022-04-09_13_08_47_713: EngineState:     3249.0 ohms -> valid: Yes. 13.8 deg C
2022-04-09_13_08_47_713: EngineState: confirmation_show_sensor 20:14
```

rusefillc, 2022-04-09:
green 0.5.2 board

```
2022-04-09_15_38_00_278: EngineState: CRITICAL error: aut2 analog input validation failed 9.880859375 vs 6.199999809 20220315@microRusEFI
2022-04-09_15_38_00_389: EngineState: CRITICAL error: aut2 analog input validation failed 9.880859375 vs 6.199999809 20220315@microRusEFI
2022-04-09_15_38_00_611: EngineState: CRITICAL error: aut2 analog input validation failed 9.880859375 vs 6.199999809 20220315@microRusEFI
2022-04-09_15_38_00_827: EngineState: CRITICAL error: aut2 analog input validation failed 9.880859375 vs 6.199999809 20220315@microRusEFI
```

let me wider acceptable range

mck1117, 2022-04-09:
That seems like a wide range, are we sure something isn't wrong? What is the test device hooked up as a "thermistor"?

rusefillc, 2022-04-09:
it's the official "text fixture" which I assume has _some_ resistors on it?

rusefillc, 2022-04-09:
master 0.5.2

```
2022-04-09_17_18_37_892: EngineState: CRITICAL error: clt analog input validation failed 66.393615722 vs 49.299999237 20220409@microRusEFI
2022-04-09_17_18_38_002: EngineState: Sensor "Clt": Raw value: 1.33 Valid: Yes Converted value 49.68
2022-04-09_17_18_38_002: EngineState:     1.33 volts -> 981.0 ohms with supply voltage 5.00 and pullup 2700.0.
2022-04-09_17_18_38_003: EngineState:     981.0 ohms -> valid: Yes. 49.6 deg C
2022-04-09_17_18_38_003: EngineState: confirmation_show_sensor 1:13
```

rusefillc, 2022-04-09:
2020 happy blue

[image]

closed 2022-04-09 by rusefillc

mck1117, 2022-04-10:
I'm trying to ask whether this was an actual hardware fault detected or not. Did you check the real voltage with a multimeter before closing this?

mck1117, 2022-04-10:
That is not an acceptable answer. Would you rather make a spot check now, or recall a whole set of units later?

rusefillc, 2022-04-10:
C'mon, there is no statistical chance of only that channel being damaged and exactly within the acceptable range.

rusefillc, 2022-04-10:
https://files.slack.com/files-pri/T4NJHQ8QZ-F03AWK8L402/image.png

commit 155b3ec8 (2022-04-09): MRE QC build broken fix #4059

nasty work-around