rusefi/rusefi #4060: Why do we need clean_env_variables.sh again?

rusefillc, 2022-04-09:
We use 'export key=value' and just 'key=value' in firmware build scripts

we also have a very weird clean_env_variables.sh

```

DEFAULT_ENGINE_TYPE=-DDEFAULT_ENGINE_TYPE=FRANKENSO_MIATA_NA6_VAF
FW_ID_ENV=frankensoNA6

export EXTRA_PARAMS="-DDUMMY \
 -DSHORT_BOARD_NAME=frankenso_na6 \
```

in the end we need all? most? variables to be consumed by common_make.sh which is always 'downstream' from the script where we set variables

Can we wipe out all 'export ' and wipe out clean_env_variables.sh as a consequence?

chuckwagoncomputing, 2022-04-09:
Variables aren't handed down to scripts that are called unless they are defined with `export`.

On the other hand, I don't know why clean_env_variables.sh exists. Every firmware build is virtually on a separate machine. Maybe it's something for/from the Jenkins workflow?

rusefillc, 2022-04-09:
So we do not expect ``FW_ID_ENV=frankensoNA6`` to work?

clean_env_variables.sh helps with consecutive executions which yes could be jenkins or even our local devices?

PR #5012 (2023-01-29): remove clean_env_variables script
In theory, this script cannot possibly do anything.
Variables set in a script only have effect within that script.
Variables set in a script with `export` only have effect within that script and any scripts or functions which that script calls.
If the script is sourced, e.g. `. clean_env_variables.sh` or `source clean_env_variables.sh`, then it would have an effect, but that's not the case here.
Fix #4060
closed 2023-01-29 by rusefillc