rusefi/rusefi #4063: Speedometer output

NMSTEC, 2022-04-10:
Issue: Car gets tach & speed data from ECU. Current SW does not seem to support VS output, short of LUA.

Please replicate Tach options into a new VS options page.

mck1117, 2022-04-10:
so use lua?

NMSTEC, 2022-04-10:
But why? It would be much easier and less memory usage than using LUA.

rusefillc, 2022-04-10:
because we have two software developers and that's one of the bottlenecks of RE :(

Legit feature request with low priority.

ufo-alert, 2022-04-19:
Is this issue something that can be reproduced using the simulator (I have no hardware)? If so, please provide any tips/hints on how to reproduce using the simulator and I'll start investigating.
-Leo

rusefillc, 2022-04-19:
Welcome @ufo-alert have you seen rusEFI slack https://rusefi.com/forum/viewtopic.php?f=13&t=1198?

This specific ticket here is not the best first time ticket. Open question what is a good first time ticket without hardware, maybe some (boring as hell) unit test coverage?