rusefi/rusefi #4084: Mark reports that detectHellenBoardType hangs up HD and Nucleo

rusefillc, 2022-04-19:
detectHellenBoardType hangs up HD hellen and Nucleo

rusefillc, 2022-06-19:
```
2022-06-19_11_28_31_739: EngineState: applyNonPersistentConfiguration()
2022-06-19_11_28_31_739: EngineState: initializeTriggerWaveform(TT_TOOTHED_WHEEL_60_2/8)
2022-06-19_11_28_31_739: EngineState: Read valid configuration from flash!
2022-06-19_11_28_31_740: EngineState: Hellen board pin states = 0 1
2022-06-19_11_28_31_740: EngineState: * Hellen 144-pin mcu detected!
2022-06-19_11_28_31_740: EngineState: Starting Hellen Board ID detection...
2022-06-19_11_28_31_741: EngineState: pin_markUsed: boardId on PF1
2022-06-19_11_28_31_741: EngineState: pin_markUnused: PF1
2022-06-19_11_28_31_741: EngineState: pin_markUsed: boardId on PF1
2022-06-19_11_28_31_742: EngineState: pin_markUnused: PF1
2022-06-19_11_28_31_742: EngineState: pin_markUsed: boardId on PF1
2022-06-19_11_28_31_742: EngineState: pin_markUnused: PF1
2022-06-19_11_28_31_742: EngineState: pin_markUsed: boardId on PF0
2022-06-19_11_28_31_743: EngineState: pin_markUnused: PF0
2022-06-19_11_28_31_743: EngineState: pin_markUsed: boardId on PF0
2022-06-19_11_28_31_743: EngineState: pin_markUnused: PF0
2022-06-19_11_28_31_744: EngineState: pin_markUsed: boardId on PF0
2022-06-19_11_28_31_744: EngineState: pin_markUnused: PF0
2022-06-19_11_28_31_744: EngineState: pin_markUnused: PF0
2022-06-19_11_28_31_745: EngineState: pin_markUnused: PF1
2022-06-19_11_28_31_745: EngineState: * RESULT: BoardId = 2, R1 = 510.000000000, R2 = 750.000000000 (Elapsed time: 306 ms)
2022-06-19_11_28_31_745: EngineState: pin_markUsed: Primary UART RX on PD5
2022-06-19_11_28_31_745: EngineState: pin_markUsed: Primary UART TX on PD6
2022-06-19_11_28_31_746: EngineState: pin_markUsed: led: comm status on PE7
2022-06-19_11_28_31_746: EngineState: pin_markUsed: led: warning status on PE8
2022-06-19_11_28_31_746: EngineState: pin_markUsed: led: running status on PG1
2022-06-19_11_28_31_747: EngineState: pin_markUsed: a-EN on 61 Reverse Gear
2022-06-19_11_28_31_747: EngineState: initAdcInputs()
2022-06-19_11_28_31_747: EngineState: pin_markUsed: SPI clock on PB13

```

rusefillc, 2022-06-19:
Anyway potential forever loop replaced with explicit firmware error

closed 2022-06-19 by rusefillc

NMSTEC, 2022-06-20:
right after fw update '''2022-06-20_15_26_58_794: ConsoleUI: COMPOSITE_OFF_RPM=300
2022-06-20_15_27_00_287: EngineState: initHardware()
2022-06-20_15_27_00_288: EngineState: pin_markUsed: led: CRITICAL status on PG0
2022-06-20_15_27_00_288: EngineState: initRtc()
2022-06-20_15_27_00_332: EngineState: engine sniffer: No
2022-06-20_15_27_00_338: EngineState: engine sniffer size=0
2022-06-20_15_27_00_341: EngineState: pin_markUsed: USB DM on PA11
2022-06-20_15_27_00_348: EngineState: pin_markUsed: USB DP on PA12
2022-06-20_15_27_00_351: EngineState: *** rusEFI LLC (c) 2012-2022. All rights reserved.
2022-06-20_15_27_00_361: EngineState: *** rusEFI v20220620@41302
2022-06-20_15_27_00_364: EngineState: *** Chibios Kernel:       6.1.4
2022-06-20_15_27_00_368: EngineState: *** Compiled:     Jun 20 2022 - 20:56:08
2022-06-20_15_27_00_371: EngineState: *** COMPILER=9.3.1 20200408 (release)
2022-06-20_15_27_00_374: EngineState: *** detected HSE clock 19.92 MHz PLLM = 20
2022-06-20_15_27_00_376: EngineState: UID=370045 46335004 20343636
2022-06-20_15_27_00_376: EngineState: can read 0x20000010 1
2022-06-20_15_27_00_377: EngineState: can read 0x20020010 1
2022-06-20_15_27_00_378: EngineState: can read 0x20070010 0
2022-06-20_15_27_00_378: EngineState: isStm32F42x Yes
2022-06-20_15_27_00_379: EngineState: MCU rev=419 flashSize=1024
2022-06-20_15_27_00_380: EngineState: CORTEX_MAX_KERNEL_PRIORITY=2
2022-06-20_15_27_00_380: EngineState: STM32_ADCCLK=21000000
2022-06-20_15_27_00_381: EngineState: STM32_TIMCLK1=84000000
2022-06-20_15_27_00_381: EngineState: STM32_TIMCLK2=168000000
2022-06-20_15_27_00_382: EngineState: STM32_PCLK1=42000000
2022-06-20_15_27_00_383: EngineState: STM32_PCLK2=84000000
2022-06-20_15_27_00_383: EngineState: PORT_IDLE_THREAD_STACK_SIZE=32
2022-06-20_15_27_00_384: EngineState: CH_DBG_ENABLE_ASSERTS=1
2022-06-20_15_27_00_385: EngineState: CH_DBG_SYSTEM_STATE_CHECK=1
2022-06-20_15_27_00_385: EngineState: CH_DBG_ENABLE_STACK_CHECK=1
2022-06-20_15_27_00_386: EngineState: EFI_LOGIC_ANALYZER=0
2022-06-20_15_27_00_386: EngineState: EFI_TUNER_STUDIO=1
2022-06-20_15_27_00_387: EngineState: EFI_SIGNAL_EXECUTOR_SLEEP=0
2022-06-20_15_27_00_388: EngineState: EFI_SIGNAL_EXECUTOR_HW_TIMER=0
2022-06-20_15_27_00_388: EngineState: EFI_SHAFT_POSITION_INPUT=1
2022-06-20_15_27_00_389: EngineState: EFI_INTERNAL_ADC=1
2022-06-20_15_27_00_389: EngineState: readFromFlash 80E0000
2022-06-20_15_27_00_390: EngineState: applyNonPersistentConfiguration()
2022-06-20_15_27_00_391: EngineState: initializeTriggerWaveform(TT_TOOTHED_WHEEL_32_2/69)
2022-06-20_15_27_00_391: EngineState: initializeTriggerWaveform(TT_ONE/18)
2022-06-20_15_27_00_392: EngineState: Read valid configuration from flash!
2022-06-20_15_27_00_393: EngineState: Hellen board pin states = 0 1
2022-06-20_15_27_00_393: EngineState: * Hellen 144-pin mcu detected!
2022-06-20_15_27_00_394: EngineState: Starting Hellen Board ID detection...
2022-06-20_15_27_00_394: EngineState: pin_markUsed: boardId on PF1
2022-06-20_15_27_00_395: EngineState: pin_markUnused: PF1
2022-06-20_15_27_00_396: EngineState: pin_markUsed: boardId on PF1
2022-06-20_15_27_00_397: EngineState: pin_markUnused: PF1
2022-06-20_15_27_00_398: EngineState: pin_markUsed: boardId on PF1
2022-06-20_15_27_00_399: EngineState: pin_markUnused: PF1
2022-06-20_15_27_00_400: EngineState: pin_markUsed: boardId on PF0
2022-06-20_15_27_00_400: EngineState: pin_markUnused: PF0
2022-06-20_15_27_00_401: EngineState: pin_markUsed: boardId on PF0
2022-06-20_15_27_00_402: EngineState: pin_markUnused: PF0
2022-06-20_15_27_00_402: EngineState: pin_markUsed: boardId on PF0
2022-06-20_15_27_00_403: EngineState: pin_markUnused: PF0
2022-06-20_15_27_00_404: EngineState: pin_markUnused: PF0
2022-06-20_15_27_00_404: EngineState: pin_markUnused: PF1
2022-06-20_15_27_00_405: EngineState: * RESULT: BoardId = 2, R1 = 510.000000000, R2 = 750.000000000 (Elapsed time: 306 ms)
2022-06-20_15_27_00_406: EngineState: pin_markUsed: Primary UART RX on PD5
2022-06-20_15_27_00_406: EngineState: pin_markUsed: Primary UART TX on PD6
2022-06-20_15_27_00_407: EngineState: pin_markUsed: led: comm status on PE7
2022-06-20_15_27_00_407: EngineState: pin_markUsed: led: warning status on PE8
2022-06-20_15_27_00_408: EngineState: pin_markUsed: led: running status on PG1
2022-06-20_15_27_00_409: EngineState: pin_markUsed: a-EN on 61 Reverse Gear
2022-06-20_15_27_00_409: EngineState: initAdcInputs()
2022-06-20_15_27_00_410: EngineState: pin_markUsed: knock ch1 on PF7
2022-06-20_15_27_00_411: EngineState: pin_markUsed: SPI clock on PB13
2022-06-20_15_27_00_411: EngineState: pin_markUsed: SPI master out on PB15
2022-06-20_15_27_00_412: EngineState: pin_markUsed: SPI master in  on PB14
[... 91 lines cut ...]
2022-06-20_15_27_00_471: EngineState: LUA: Compression release solenoid 0
2022-06-20_15_27_00_471: EngineState: LUA: Compression release solenoid 0
2022-06-20_15_27_00_472: EngineState: USB: got Query command
2022-06-20_15_27_00_472: EngineState: got S/H (queryCommand) mode=1
2022-06-20_15_27_00_473: EngineState: Primary UART RX
2022-06-20_15_27_00_473: EngineState: Primary UART TX
2022-06-20_15_27_00_474: EngineState: TunerStudio size=616 / total=3 / errors=0 / H=1 / O=0 / P=0 / B=0
2022-06-20_15_27_00_475: EngineState: TunerStudio W=0 / C=0 / P=0
2022-06-20_15_27_00_476: EngineState: USB: got Query command
2022-06-20_15_27_00_476: EngineState: got S/H (queryCommand) mode=1
2022-06-20_15_27_00_477: EngineState: Primary UART RX
[... cut ...]
2022-06-20_15_27_00_479: EngineState: READ mode=1 offset=1000 size=500
2022-06-20_15_27_00_480: EngineState: READ mode=1 offset=1500 size=500
2022-06-20_15_27_00_480: EngineState: READ mode=1 offset=2000 size=500
2022-06-20_15_27_00_481: EngineState: READ mode=1 offset=2500 size=500
2022-06-20_15_27_00_481: EngineState: READ mode=1 
2022-06-20_15_27_00_558: EngineState: 0 - unix time
2022-06-20_15_27_00_559: EngineState: Current RTC localtime is: 2022-06-20 15:27:00 w=1
2022-06-20_15_27_00_561: EngineState: confirmation_set date 1655738820:19
'''

NMSTEC, 2022-06-20:
After update, trying to update Tuner Studio settings results in error: 
[image]

rusefillc, 2022-06-21:
```
2022-06-20_15_27_00_405: EngineState: * RESULT: BoardId = 2, R1 = 510.000000000, R2 = 750.000000000 (Elapsed time: 306 ms)
```

hellen board ID alive

everything else not relevant

commit 50a8508b (2022-06-19): Mark reports that detectHellenBoardType hangs up HD and Nucleo fix #4084