rusefi/rusefi #4087: VSS issue

rusefillc, 2022-04-20:
@chuckwagoncomputing reports https://youtu.be/jUPMh9Ew7Sk

chuckwagoncomputing, 2022-04-20:
Another demonstration: https://youtu.be/pnZfQMugd1U

chuckwagoncomputing, 2022-04-20:
Once when I was messing around with it I put an efiPrintf() in the converter function for input shaft speed, and after going really high and back down as I did in the second video, the frequency parameter being passed to it was alternating between a very high number and a very negative number. Not sure how you can have negative frequency.

chuckwagoncomputing, 2022-04-20:
Now I tried butting an efiPrintf in the VSS converter function, and when I turn on the generator after the board, the converter function is getting called, but with a frequency of 0.

chuckwagoncomputing, 2022-04-22:
Frequency before the call to filter() is non-zero, and after is zero.
https://github.com/rusefi/rusefi/blob/786ec5a4ddb3adc6c5cc4f9e4a6fd2b796cae762/firmware/controllers/sensors/frequency_sensor.cpp#L57

mck1117, 2022-04-22:
what was the filter parameter?

chuckwagoncomputing, 2022-04-22:
500mS

chuckwagoncomputing, 2022-04-22:
Oh that's the timeout period - I tried with a wide range of filter parameters and it didn't seem to make any difference.

chuckwagoncomputing, 2022-04-28:
Looks like Matt's PR has solved this!
Basically (as I understand it) the filter parameter was being ignored, and a very bad (low) filter parameter was being used, which led to some unworkable floating point math.

closed 2022-04-28 by chuckwagoncomputing