rusefi/rusefi #4088: Kasper reports issue with lua ETB adder

rusefillc, 2022-04-20:


mck1117, 2022-04-20:
what does issue mean?

rusefillc, 2022-04-21:
without etbAdd
```
function onTick()
	pps = getSensor("AcceleratorPedal")
	tps = getSensor("tps1")

	print (" PPS: " ..pps .." TPS: " ..tps)

	-- just something silly for test
	-- 	setEtbAdd(pps)
end
```

output
```
2022-04-20_21_31_37_514: EngineState: LUA:  PPS: 35.1543960 TPS: 40.7413635
2022-04-20_21_31_37_623: EngineState: LUA:  PPS: 34.1104660 TPS: 37.2634887
2022-04-20_21_31_37_734: EngineState: LUA:  PPS: 33.8998718 TPS: 36.3833312
2022-04-20_21_31_37_843: EngineState: LUA:  PPS: 34.2650184 TPS: 35.6362228
2022-04-20_21_31_37_953: EngineState: LUA:  PPS: 34.4048500 TPS: 35.7102165
2022-04-20_21_31_38_060: EngineState: LUA:  PPS: 34.3905410 TPS: 35.8502120
2022-04-20_21_31_38_171: EngineState: LUA:  PPS: 34.3647232 TPS: 35.9551086
2022-04-20_21_31_38_280: EngineState: LUA:  PPS: 34.3445587 TPS: 36.0940246
2022-04-20_21_31_38_388: EngineState: LUA:  PPS: 34.2466125 TPS: 36.2892074
2022-04-20_21_31_38_388: EngineState: LUA:  PPS: 31.7630424 TPS: 36.2192115
2022-04-20_21_31_38_497: EngineState: LUA:  PPS: 28.7460498 TPS: 33.7411193
2022-04-20_21_31_38_607: EngineState: LUA:  PPS: 24.6745376 TPS: 28.6628265
2022-04-20_21_31_38_717: EngineState: LUA:  PPS: 21.5463809 TPS: 23.9635429
2022-04-20_21_31_38_826: EngineState: LUA:  PPS: 19.3132610 TPS: 20.2152309
2022-04-20_21_31_38_935: EngineState: LUA:  PPS: 16.8730525 TPS: 11.5644893
2022-04-20_21_31_39_043: EngineState: LUA:  PPS: 16.8370952 TPS: 13.0407886
2022-04-20_21_31_39_151: EngineState: LUA:  PPS: 16.8387069 TPS: 14.9069442
2022-04-20_21_31_39_261: EngineState: LUA:  PPS: 16.8386993 TPS: 14.9944086
2022-04-20_21_31_39_370: EngineState: LUA:  PPS: 16.8507881 TPS: 15.0321311
2022-04-20_21_31_39_478: EngineState: LUA:  PPS: 16.8707389 TPS: 15.0477294
2022-04-20_21_31_39_589: EngineState: LUA:  PPS: 18.8952579 TPS: 16.0394897
2022-04-20_21_31_39_589: EngineState: LUA:  PPS: 22.5859527 TPS: 30.1832084
2022-04-20_21_31_39_695: EngineState: LUA:  PPS: 25.0368232 TPS: 31.8815078
2022-04-20_21_31_39_805: EngineState: LUA:  PPS: 25.5592575 TPS: 31.2820854
2022-04-20_21_31_39_915: EngineState: LUA:  PPS: 30.3257980 TPS: 31.3790569
2022-04-20_21_31_40_025: EngineState: LUA:  PPS: 31.9933223 TPS: 32.7483558
2022-04-20_21_31_40_134: EngineState: LUA:  PPS: 29.4365196 TPS: 33.5595893
2022-04-20_21_31_40_242: EngineState: LUA:  PPS: 26.0332164 TPS: 31.4847373
2022-04-20_21_31_40_352: EngineState: LUA:  PPS: 22.8130035 TPS: 26.6076107
2022-04-20_21_31_40_461: EngineState: LUA:  PPS: 21.0398254 TPS: 22.8692436
2022-04-20_21_31_40_568: EngineState: LUA:  PPS: 28.1545982 TPS: 25.4460334
2022-04-20_21_31_40_679: EngineState: LUA:  PPS: 35.9546051 TPS: 36.7301177
2022-04-20_21_31_40_788: EngineState: LUA:  PPS: 38.5535354 TPS: 41.1739654
2022-04-20_21_31_40_789: EngineState: LUA:  PPS: 39.4399337 TPS: 42.0024185
2022-04-20_21_31_40_898: EngineState: LUA:  PPS: 34.7178993 TPS: 41.4086227
2022-04-20_21_31_41_007: EngineState: LUA:  PPS: 27.6497611 TPS: 31.9098663
2022-04-20_21_31_41_118: EngineState: LUA:  PPS: 31.2879791 TPS: 30.7178649
2022-04-20_21_31_41_227: EngineState: LUA:  PPS: 40.5082931 TPS: 39.4750823
2022-04-20_21_31_41_336: EngineState: LUA:  PPS: 45.2098655 TPS: 46.2981185
2022-04-20_21_31_41_447: EngineState: LUA:  PPS: 48.2619171 TPS: 50.1188011
2022-04-20_21_31_41_558: EngineState: LUA:  PPS: 48.3266983 TPS: 50.6287612
2022-04-20_21_31_41_668: EngineState: LUA:  PPS: 39.1217308 TPS: 47.3720245
2022-04-20_21_31_41_773: EngineState: LUA:  PPS: 28.0209999 TPS: 31.9926605

```

rusefillc, 2022-04-21:
with etbAdd
```
function onTick()
	pps = getSensor("AcceleratorPedal")
	tps = getSensor("tps1")

	print (" PPS: " ..pps .." TPS: " ..tps)

	-- just something silly for test
	setEtbAdd(pps)
end
```

обрати вниманием, как TSP уходит от PPS
```
2022-04-20_21_33_22_116: EngineState: LUA:  PPS: 0.1699165 TPS: 3.5934977
2022-04-20_21_33_22_228: EngineState: LUA:  PPS: 0.1736958 TPS: 3.5901007
2022-04-20_21_33_22_335: EngineState: LUA:  PPS: 0.1770435 TPS: 3.5969462
2022-04-20_21_33_22_443: EngineState: LUA:  PPS: 0.2037015 TPS: 3.6005494
2022-04-20_21_33_22_551: EngineState: LUA:  PPS: 0.2106136 TPS: 3.5960912
2022-04-20_21_33_22_659: EngineState: LUA:  PPS: 0.2641530 TPS: 3.5970609
2022-04-20_21_33_22_770: EngineState: LUA:  PPS: 1.0565030 TPS: 3.6129338
2022-04-20_21_33_22_881: EngineState: LUA:  PPS: 2.2318325 TPS: 4.7390074
2022-04-20_21_33_22_990: EngineState: LUA:  PPS: 3.5141167 TPS: 5.9913029
2022-04-20_21_33_23_102: EngineState: LUA:  PPS: 4.8694067 TPS: 6.0970978
2022-04-20_21_33_23_102: EngineState: LUA:  PPS: 7.2981796 TPS: 9.1755046
2022-04-20_21_33_23_209: EngineState: LUA:  PPS: 9.3942565 TPS: 13.6827440
2022-04-20_21_33_23_317: EngineState: LUA:  PPS: 9.9058647 TPS: 22.3335609
2022-04-20_21_33_23_426: EngineState: LUA:  PPS: 11.3994922 TPS: 28.0363883
2022-04-20_21_33_23_535: EngineState: LUA:  PPS: 14.1494655 TPS: 30.3225364
2022-04-20_21_33_23_644: EngineState: LUA:  PPS: 18.0731468 TPS: 35.7017211
2022-04-20_21_33_23_751: EngineState: LUA:  PPS: 21.2294883 TPS: 42.8243713
2022-04-20_21_33_23_864: EngineState: LUA:  PPS: 22.1967849 TPS: 47.3526420
2022-04-20_21_33_23_973: EngineState: LUA:  PPS: 23.6303939 TPS: 48.6481704
2022-04-20_21_33_24_084: EngineState: LUA:  PPS: 24.0710182 TPS: 50.5723915
2022-04-20_21_33_24_194: EngineState: LUA:  PPS: 24.8026981 TPS: 51.2551269
```

вывод: у меня настоящий тест на столе работает с сегодняшней прошивкой

closed 2022-04-21 by rusefillc