rusefi/rusefi #4091: Table rpm values non linear

BeerMoneyMotorsports, 2022-04-22:
While reviewing a tune, it was found that th rpm values in the table were not linear.  This gave no errors to the user. 

https://rusefi.com/online/view.php?msq=963

[image]

rusefillc, 2022-04-22:
@BeerMoneyMotorsports any chance that you mean "ascending"?

rusefillc, 2022-04-22:
We are invoking
```
		ensureArrayIsAscending("Ignition load", config->ignitionLoadBins);
		ensureArrayIsAscending("Ignition RPM", config->ignitionRpmBins);
```
on boot not every time properties were changed.

@BeerMoneyMotorsports please confirm that ECU was power cycled with those settings

rusefillc, 2022-04-23:
@BeerMoneyMotorsports is this a tune from someone or can you reproduce this locally? meaning to be 100% sure on sequence of events?

mck1117, 2022-04-23:
I loaded this tune on random recent firmware and it errored

rusefillc, 2022-04-30:
@BeerMoneyMotorsports do you know steps to reproduce?

rusefillc, 2022-05-08:
no response from OP

closed 2022-05-08 by rusefillc