rusefi/rusefi #4100: no cranking fuel *lemon issue

BeerMoneyMotorsports, 2022-04-23:
when moving from older firmware(1/24) to latest release(3/16) the lemon presents a no start symptom.

Observations- 
1- during cranking, there is only one flash of all 4 injectors (simultaneous) 
2- after being warmed up, it will start on the newer firmware.  It is noted that the single simultaneous injection is enough for it to exit "cranking" and enter "running" mode.  This is clear by watching the injector LEDs on the board. 
3- If you raise the "cranking" rpm limit higher, it will attempt to start on that single simultaneous injection but will never enter "running" injection mode.  This results again in the no start.
4- with 1/24 firmware, you can several quick simultaneous pulses from the injectors prior to entering "running" injection mode.

Log with 1/24 firmware
https://rusefi.com/online/view.php?log=758

Log with 3/16 firmware
https://rusefi.com/online/view.php?log=757

Something in the new firmware is preventing the ecu from injection during cranking.

rusefillc, 2022-04-23:
@BeerMoneyMotorsports we have a LOT more logging in snapshot version / Release candidate

please use "view -> engine state" in release candidate to have live view

```
	liveGraph = engine_state_4_Graph, "Graph", South
		graphLine = injectorFlowCorrection
		graphLine = baroCorrection
		graphLine = cranking_baseFuel
		graphLine = cranking_coolantTemperatureCoefficient
	liveGraph = engine_state_5_Graph, "Graph", South
		graphLine = cranking_tpsCoefficient
		graphLine = cranking_durationCoefficient
		graphLine = cranking_fuel
		graphLine = running_postCrankingFuelCorrection

```

BeerMoneyMotorsports, 2022-04-24:
log from 4/24 firmware.  This firmware seems to function opposite of 3/16.  The injector leds stayed on solid with some flashing and has flooded the engine on "cold" start.  

https://rusefi.com/online/view.php?log=759

BeerMoneyMotorsports, 2022-04-24:
Tested with the e85 cranking table on but set to zero as shown below.  Log shows Last Injections in line with the 1/24 firmware.  Car starts normally.

https://rusefi.com/online/view.php?log=760

[image]

rusefillc, 2022-04-24:
Progress!!!

BeerMoneyMotorsports, 2022-04-25:
logs of various values in the E85 cranking table showing behavior inconsistent with expectations.

All Multiplier 0
https://rusefi.com/online/view.php?log=761

All Multiplier 0.02
https://rusefi.com/online/view.php?log=762

All Multiplier 0.2
https://rusefi.com/online/view.php?log=763

All Multiplier 1.0
https://rusefi.com/online/view.php?log=764

mck1117, 2022-04-25:
I will take a look at this tomorrow for sure

BeerMoneyMotorsports, 2022-04-25:
Any thoughts?  A misplaced decimal maybe?

mck1117, 2022-04-26:
it's a missed `autoscale`

rusefillc, 2022-04-26:
@mck1117 do we have test coverage here?

PR #4105 (2022-04-26): missed e85 cranking autoscale
fix #4100
closed 2022-04-26 by rusefillc