rusefi/rusefi #4132: Corolla 1ZZ 2003 trigger improvements

rusefillc, 2022-05-02:
[2022-05-02_14.12.19.zip](https://github.com/rusefi/rusefi/files/8605596/2022-05-02_14.12.19.zip)

[image]

rusefillc, 2022-05-02:
[image]

should be 

[image]

rusefillc, 2022-05-02:
@eldominio please confirm that ^^^ setting have helped

Shall we implement this as new official hard-coded trigger?

mck1117, 2022-05-02:
I don't think we need it as an official hardcoded trigger, I think we can just hand tune the ratios for all 36-2 triggers. Those overrides shouldn't hurt anybody else using a 36-1 on any engine.

rusefillc, 2022-05-02:
@mck1117 so you say do not ADD new 36-2 but change current 36-2?

ElDominio, 2022-05-02:
I would say just call it Toyota 36-2, since 1ZZ, 2ZZ, 2AR, 2ZR and pretty much all 36-2 4cylinder Toyotas use it at even the same angle. BTW Trigger Angle is -140.

And please rename triggers with "/" to + or - pleeeease

mck1117, 2022-05-02:
> @mck1117 so you say do not ADD new 36-2 but change current 36-2?

correct

closed 2022-05-04 by rusefillc

rusefillc, 2022-05-13:
Usability follow up https://github.com/rusefi/rusefi/issues/4172

commit 7547ed77 (2022-05-04): Corolla 1ZZ 2003 trigger improvements fix #4132
commit efccb210 (2022-05-04): Corolla 1ZZ 2003 trigger improvements fix #4132
PR #4138 (2022-05-05): fix 36-2 trigger params
The problems were with 36-2, not 36-1....

fix #4132 again