rusefi/rusefi #4134: MRE wasted spark config error

OrchardPerformance, 2022-05-03:
Looks like the TS config for the MRE is highlighting the wrong box. 
Confirmed on previous release (20220126) but not had a chance to check the newer releases yet. 

I don't think it is causing a running issue by setting Ignition 7 as output 4 but trying to confirm this currently.

[image]

OrchardPerformance, 2022-05-03:
Extra context on the firing order 
[image]

OrchardPerformance, 2022-05-03:
Tune here for completeness. 

https://rusefi.com/online/view.php?msq=989

OrchardPerformance, 2022-05-03:
Further investigation shows we actually highlight only the first 4 cylinders of the firing order. 
This is consistent behavior for all V8 firing orders available on MRE but is it the best/right way to do this? 
[image]
[image]

rusefillc, 2022-05-04:
@OrchardPerformance this was not touched for a few years

yes always first half, why does this look wrong to you?

OrchardPerformance, 2022-08-29:
It's not wrong, just different, possibly quite clever but not currently well explained to tuners familiar with other systems). 

Closing this one off as there is no software problem, maybe just a documentation one.

closed 2022-08-29 by OrchardPerformance