rusefi/rusefi #4163: Leduce flash footprint by smarter code generation

rusefillc, 2022-05-11:
OpenBLT does not fit inside 512K flash on F7 with work-around this points us at ``value_lookup_generated.cpp``

[image]

rusefillc, 2022-05-11:
base line
```
/home/runner/work/rusefi/rusefi/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: build/rusefi.elf section `.rodata' will not fit in region `flash0'
/home/runner/work/rusefi/rusefi/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: region `flash0' overflowed by 16680 bytes
collect2: error: ld returned 1 exit status
make: *** [ChibiOS/os/common/startup/ARMCMx/compilers/GCC/mk/rules.mk:217: build/rusefi.elf] Error 1
ERROR invoking compile_proteus_f7.sh
Error: Process completed with exit code 1
```

rusefillc, 2022-12-04:
@mck1117 at least string literals are gone, is it good enough?

mck1117, 2022-12-05:
I wish we'd done this in a branch so we could actually measure the impact 😢

rusefillc, 2022-12-05:
@mck1117 we have the RC binaries from few days ago

mck1117, 2022-12-05:
well, backporting current jar to the last commit before this latest batch of changes:

old lookup: 639054
new lookup: 636846
delta: 2208 bytes saved

So, did slightly save space! However I think there's a much bigger performance improvement, since its now comparing integers instead of the full string on every line.

closed 2022-12-05 by mck1117