rusefi/rusefi #4178: Custom Toothed Wheel position should be able to be selected

ElDominio, 2022-05-16:
[image]

We frequently run into 24-1 modified Toyota triggers on cam, or 4 teeth without missing. New update doesn't let us do this since the option is grayed out

mck1117, 2022-05-16:
As a temporary workaround, you can select 60-2, set "on camshaft", then put it back to the correct trigger. The ECU is looking at those settings even though you can't change them in TS.

rusefillc, 2022-09-05:
Looks like this fixed at some point - I did not do anything and it's selectable for me now

[image]

closed 2022-09-05 by rusefillc