rusefi/rusefi #4191: validation of already initialized BT

rusefillc, 2022-05-20:
What if we initialize BT module once

What's expected behavior after that? How can we distinguish broken BT from already programmed working BT?

```
2022-05-19_22_22_54_536: EngineState: Current RTC localtime is: 2022-05-19 22:22:54 w=1
2022-05-19_22_22_54_538: EngineState: confirmation_set date 1652998974:19
2022-05-19_22_22_56_119: EngineState: confirmation_bluetooth_jdy 115200 alphax 1234:32
2022-05-19_22_22_56_120: EngineState: *** Bluetooth module setup procedure ***
2022-05-19_22_22_56_120: EngineState: !Warning! Please make sure you're not currently using the BT module for communication (not paired)!
2022-05-19_22_22_56_121: EngineState: TO START THE PROCEDURE: PLEASE DISCONNECT YOUR PC COM-PORT FROM THE BOARD NOW!
2022-05-19_22_22_56_126: EngineState: After that please don't turn off the board power and wait for ~15 seconds to complete. Then reconnect to the board!
2022-05-19_22_22_56_341: EngineState: Sleeping...
2022-05-19_22_22_57_326: EngineState: Restarting at 9600
2022-05-19_22_22_57_437: EngineState: Sending AT+VERSION
2022-05-19_22_22_58_421: EngineState: Running at 9600
2022-05-19_22_22_58_422: EngineState: Sending AT+BAUD8
2022-05-19_22_22_58_535: EngineState: received 2 byte(s) 1 [+][O]
2022-05-19_22_22_59_523: EngineState: Restarting at 115200
2022-05-19_22_22_59_524: EngineState: Sending AT+NAMEalphax
2022-05-19_22_22_59_630: EngineState: received 2 byte(s) 1 [+][O]
2022-05-19_22_23_00_629: EngineState: Running at 115200
2022-05-19_22_23_00_629: EngineState: Sending AT+PIN1234
2022-05-19_22_23_01_613: EngineState: Restarting at 38400
2022-05-19_22_23_01_614: EngineState: SUCCESS! All commands (4 of 4) passed to the Bluetooth module!
2022-05-19_22_25_45_021: EngineState: confirmation_bluetooth_jdy 115200 alphax 1234:32
2022-05-19_22_25_45_022: EngineState: *** Bluetooth module setup procedure ***
2022-05-19_22_25_45_022: EngineState: !Warning! Please make sure you're not currently using the BT module for communication (not paired)!
2022-05-19_22_25_45_022: EngineState: TO START THE PROCEDURE: PLEASE DISCONNECT YOUR PC COM-PORT FROM THE BOARD NOW!
2022-05-19_22_25_45_022: EngineState: After that please don't turn off the board power and wait for ~15 seconds to complete. Then reconnect to the board!
2022-05-19_22_25_45_682: EngineState: Sleeping...
2022-05-19_22_25_46_660: EngineState: Restarting at 9600
2022-05-19_22_25_46_662: EngineState: Sending AT+VERSION
2022-05-19_22_25_47_646: EngineState: received 0 byte(s) 0 [U][U]
2022-05-19_22_25_48_741: EngineState: Restarting at 38400
2022-05-19_22_25_48_741: EngineState: Sending AT+VERSION
2022-05-19_22_25_49_729: EngineState: received 0 byte(s) 0 [U][U]
2022-05-19_22_25_50_710: EngineState: Restarting at 115200
2022-05-19_22_25_50_710: EngineState: Sending AT+VERSION
2022-05-19_22_25_51_693: EngineState: received 0 byte(s) 0 [U][U]
2022-05-19_22_25_52_721: EngineState: Restarting at 57600
2022-05-19_22_25_52_721: EngineState: Sending AT+VERSION
2022-05-19_22_25_53_780: EngineState: received 0 byte(s) 0 [U][U]
2022-05-19_22_25_54_747: EngineState: Restarting at 19200
2022-05-19_22_25_54_747: EngineState: Sending AT+VERSION
2022-05-19_22_25_55_723: EngineState: received 0 byte(s) 0 [U][U]
2022-05-19_22_25_56_816: EngineState: Restarting at 4800
2022-05-19_22_25_56_816: EngineState: Sending AT+VERSION
2022-05-19_22_25_57_795: EngineState: received 0 byte(s) 0 [U][U]
2022-05-19_22_25_58_782: EngineState: Restarting at 2400
2022-05-19_22_25_58_782: EngineState: Sending AT+VERSION
2022-05-19_22_25_59_874: EngineState: received 0 byte(s) 0 [U][U]
2022-05-19_22_26_00_858: EngineState: Restarting at 1200
2022-05-19_22_26_00_859: EngineState: Sending AT+VERSION
2022-05-19_22_26_01_948: EngineState: received 0 byte(s) 0 [U][U]
2022-05-19_22_26_02_933: EngineState: FAIL! Only 0 commands (of 4 total) were passed to the Bluetooth module!

```

rusefillc, 2022-11-16:
https://github.com/rusefi/rusefi/pull/4653

rusefillc, 2022-11-17:
@dron0gus current state

on a board of unknown quality

```
2022-11-16_19_54_30_433: EngineState: confirmation_bluetooth_jdy 115200 alphax 1234:32
2022-11-16_19_54_30_433: EngineState: *** Bluetooth module setup procedure ***
2022-11-16_19_54_30_434: EngineState: !Warning! Please make sure you're not currently using the BT module for communication (not paired)!
2022-11-16_19_54_30_434: EngineState: TO START THE PROCEDURE: PLEASE DISCONNECT YOUR PC COM-PORT FROM THE BOARD NOW!
2022-11-16_19_54_30_434: EngineState: After that please don't turn off the board power and wait for ~15 seconds to complete. Then reconnect to the board!
2022-11-16_19_54_31_314: EngineState: Sleeping...
2022-11-16_19_54_32_298: EngineState: Restarting at 9600
2022-11-16_19_54_32_298: EngineState: Sending AT+VERSION  
2022-11-16_19_54_33_283: EngineState: timeout receiving BT reply
2022-11-16_19_54_34_376: EngineState: Restarting at 38400
2022-11-16_19_54_34_376: EngineState: Sending AT+VERSION  
2022-11-16_19_54_35_355: EngineState: timeout receiving BT reply
2022-11-16_19_54_36_345: EngineState: Restarting at 115200
2022-11-16_19_54_36_345: EngineState: Sending AT+VERSION  
2022-11-16_19_54_36_346: EngineState: received 2 byte(s) 0 [ ][5]
2022-11-16_19_54_37_440: EngineState: Restarting at 57600
2022-11-16_19_54_37_442: EngineState: Sending AT+VERSION  
2022-11-16_19_54_38_421: EngineState: timeout receiving BT reply
2022-11-16_19_54_39_406: EngineState: Restarting at 19200
2022-11-16_19_54_39_407: EngineState: Sending AT+VERSION  
2022-11-16_19_54_40_393: EngineState: timeout receiving BT reply
2022-11-16_19_54_41_488: EngineState: Restarting at 4800
2022-11-16_19_54_41_488: EngineState: Sending AT+VERSION  
2022-11-16_19_54_42_470: EngineState: timeout receiving BT reply
2022-11-16_19_54_43_457: EngineState: Restarting at 2400
2022-11-16_19_54_43_458: EngineState: Sending AT+VERSION  
2022-11-16_19_54_43_568: EngineState: received 2 byte(s) 0 [ ][ ]
2022-11-16_19_54_44_549: EngineState: Restarting at 1200
2022-11-16_19_54_44_549: EngineState: Sending AT+VERSION  
2022-11-16_19_54_44_660: EngineState: received 2 byte(s) 0 [ ][ ]
2022-11-16_19_54_45_644: EngineState: FAIL! Only 0 commands (of 4 total) were passed to the Bluetooth module!
```

rusefillc, 2022-11-17:
i.e. something is NOT good with master

i will need to flash some boards in about a week so might end up reverting https://github.com/rusefi/rusefi/pull/4653

rusefillc, 2022-11-17:
(comments above removed since PROBABLY made on master without the change)

rusefillc, 2022-11-17:
board with BT already programmed: note different response at 115K.

expected: something positive
actual result: sad

```
2022-11-17_00_06_24_010: EngineState: confirmation_bluetooth_jdy 115200 alphax 1234:32
2022-11-17_00_06_24_011: EngineState: *** Bluetooth module setup procedure ***
2022-11-17_00_06_24_011: EngineState: !Warning! Please make sure you're not currently using the BT module for communication (not paired)!
2022-11-17_00_06_24_011: EngineState: TO START THE PROCEDURE: PLEASE DISCONNECT YOUR PC COM-PORT FROM THE BOARD NOW!
2022-11-17_00_06_24_011: EngineState: After that please don't turn off the board power and wait for ~15 seconds to complete. Then reconnect to the board!
2022-11-17_00_06_24_678: EngineState: Sleeping...
2022-11-17_00_06_25_777: EngineState: Restarting at 9600
2022-11-17_00_06_25_778: EngineState: Sending AT+VERSION  
2022-11-17_00_06_26_763: EngineState: timeout receiving BT reply
2022-11-17_00_06_27_747: EngineState: Restarting at 38400
2022-11-17_00_06_27_748: EngineState: Sending AT+VERSION  
2022-11-17_00_06_28_838: EngineState: timeout receiving BT reply
2022-11-17_00_06_29_819: EngineState: Restarting at 115200
2022-11-17_00_06_29_820: EngineState: Sending AT+VERSION  
2022-11-17_00_06_29_820: EngineState: received 2 byte(s) 0 [\][ ]
2022-11-17_00_06_30_806: EngineState: Restarting at 57600
2022-11-17_00_06_30_806: EngineState: Sending AT+VERSION  
2022-11-17_00_06_31_791: EngineState: timeout receiving BT reply
2022-11-17_00_06_32_886: EngineState: Restarting at 19200
2022-11-17_00_06_32_886: EngineState: Sending AT+VERSION  
2022-11-17_00_06_33_871: EngineState: timeout receiving BT reply
2022-11-17_00_06_34_858: EngineState: Restarting at 4800
2022-11-17_00_06_34_858: EngineState: Sending AT+VERSION  
2022-11-17_00_06_35_950: EngineState: timeout receiving BT reply
2022-11-17_00_06_36_933: EngineState: Restarting at 2400
2022-11-17_00_06_36_933: EngineState: Sending AT+VERSION  
2022-11-17_00_06_37_046: EngineState: received 2 byte(s) 0 [ ][ ]
2022-11-17_00_06_38_043: EngineState: Restarting at 1200
2022-11-17_00_06_38_044: EngineState: Sending AT+VERSION  
2022-11-17_00_06_38_139: EngineState: received 2 byte(s) 0 [ ][ ]
2022-11-17_00_06_39_122: EngineState: FAIL! Only 0 commands (of 4 total) were passed to the Bluetooth module!

```

rusefillc, 2022-11-17:
another board status unclear

```
2022-11-17_00_10_08_924: EngineState: confirmation_bluetooth_jdy 115200 alphax 1234:32
2022-11-17_00_10_08_924: EngineState: *** Bluetooth module setup procedure ***
2022-11-17_00_10_08_925: EngineState: !Warning! Please make sure you're not currently using the BT module for communication (not paired)!
2022-11-17_00_10_08_925: EngineState: TO START THE PROCEDURE: PLEASE DISCONNECT YOUR PC COM-PORT FROM THE BOARD NOW!
2022-11-17_00_10_08_926: EngineState: After that please don't turn off the board power and wait for ~15 seconds to complete. Then reconnect to the board!
2022-11-17_00_10_09_696: EngineState: Sleeping...
2022-11-17_00_10_10_788: EngineState: Restarting at 9600
2022-11-17_00_10_10_788: EngineState: Sending AT+VERSION  
2022-11-17_00_10_11_788: EngineState: timeout receiving BT reply
2022-11-17_00_10_12_758: EngineState: Restarting at 38400
2022-11-17_00_10_12_759: EngineState: Sending AT+VERSION  
2022-11-17_00_10_13_850: EngineState: timeout receiving BT reply
2022-11-17_00_10_14_839: EngineState: Restarting at 115200
2022-11-17_00_10_14_841: EngineState: Sending AT+VERSION  
2022-11-17_00_10_14_841: EngineState: received 2 byte(s) 0 [ ][5]
2022-11-17_00_10_15_826: EngineState: Restarting at 57600
2022-11-17_00_10_15_826: EngineState: Sending AT+VERSION  
2022-11-17_00_10_16_924: EngineState: timeout receiving BT reply
2022-11-17_00_10_17_908: EngineState: Restarting at 19200
2022-11-17_00_10_17_909: EngineState: Sending AT+VERSION  
2022-11-17_00_10_18_887: EngineState: timeout receiving BT reply
2022-11-17_00_10_19_873: EngineState: Restarting at 4800
2022-11-17_00_10_19_874: EngineState: Sending AT+VERSION  
2022-11-17_00_10_20_967: EngineState: timeout receiving BT reply
2022-11-17_00_10_21_952: EngineState: Restarting at 2400
2022-11-17_00_10_21_952: EngineState: Sending AT+VERSION  
2022-11-17_00_10_22_063: EngineState: received 2 byte(s) 0 [ ][ ]
2022-11-17_00_10_23_057: EngineState: Restarting at 1200
2022-11-17_00_10_23_058: EngineState: Sending AT+VERSION  
2022-11-17_00_10_23_166: EngineState: received 2 byte(s) 0 [ ][ ]
2022-11-17_00_10_24_153: EngineState: FAIL! Only 0 commands (of 4 total) were passed to the Bluetooth module!

```

@dron0gus you have a 4chan board now can you test with master?

dron0gus, 2022-11-17:
@rusefillc please note that you should not see following messages for jdy-33 (https://github.com/rusefi/rusefi/blob/master/firmware/console/binary/bluetooth.cpp#L239)
```
2022-11-17_00_10_08_925: EngineState: !Warning! Please make sure you're not currently using the BT module for communication (not paired)!
2022-11-17_00_10_08_925: EngineState: TO START THE PROCEDURE: PLEASE DISCONNECT YOUR PC COM-PORT FROM THE BOARD NOW!
2022-11-17_00_10_08_926: EngineState: After that please don't turn off the board power and wait for ~15 seconds to complete. Then reconnect to the board!
```
Please make sure you are running latest-greatest build.

rusefillc, 2022-11-17:
@dron0gus 

```
2022-11-17_12_08_05_108: EngineState: confirmation_bluetooth_jdy 115200 alphax 1234:32
2022-11-17_12_08_05_108: EngineState: *** Bluetooth module setup procedure ***
2022-11-17_12_08_05_109: EngineState: Sleeping...
2022-11-17_12_08_06_202: EngineState: Restarting at 115200
2022-11-17_12_08_06_203: EngineState: sending AT  
2022-11-17_12_08_06_654: EngineState: Timeout waiting for BT reply
2022-11-17_12_08_06_654: EngineState: sending AT+DISC  
2022-11-17_12_08_07_189: EngineState: Timeout waiting for BT reply
2022-11-17_12_08_07_191: EngineState: Restarting at 9600
2022-11-17_12_08_07_192: EngineState: sending AT  
2022-11-17_12_08_07_736: EngineState: Timeout waiting for BT reply
2022-11-17_12_08_07_737: EngineState: sending AT+DISC  
2022-11-17_12_08_08_173: EngineState: Timeout waiting for BT reply
2022-11-17_12_08_08_174: EngineState: Restarting at 38400
2022-11-17_12_08_08_282: EngineState: sending AT  
2022-11-17_12_08_08_721: EngineState: Timeout waiting for BT reply
2022-11-17_12_08_08_722: EngineState: sending AT+DISC  
2022-11-17_12_08_09_262: EngineState: Timeout waiting for BT reply
2022-11-17_12_08_09_264: EngineState: Restarting at 2400
2022-11-17_12_08_09_264: EngineState: sending AT  
2022-11-17_12_08_09_815: EngineState: Timeout waiting for BT reply
2022-11-17_12_08_09_816: EngineState: sending AT+DISC  
2022-11-17_12_08_10_359: EngineState: Timeout waiting for BT reply
2022-11-17_12_08_10_359: EngineState: Restarting at 4800
2022-11-17_12_08_10_360: EngineState: sending AT  
2022-11-17_12_08_10_908: EngineState: Timeout waiting for BT reply
2022-11-17_12_08_10_909: EngineState: sending AT+DISC  
2022-11-17_12_08_11_346: EngineState: Timeout waiting for BT reply
2022-11-17_12_08_11_347: EngineState: Restarting at 19200
2022-11-17_12_08_11_454: EngineState: sending AT  
2022-11-17_12_08_11_890: EngineState: Timeout waiting for BT reply
2022-11-17_12_08_11_890: EngineState: sending AT+DISC  
2022-11-17_12_08_12_450: EngineState: Timeout waiting for BT reply
2022-11-17_12_08_12_450: EngineState: Restarting at 57600
2022-11-17_12_08_12_450: EngineState: sending AT  
2022-11-17_12_08_12_980: EngineState: Timeout waiting for BT reply
2022-11-17_12_08_12_981: EngineState: sending AT+DISC  
2022-11-17_12_08_13_418: EngineState: Timeout waiting for BT reply
2022-11-17_12_08_13_419: EngineState: Failed to find current BT module baudrate

```

dron0gus, 2022-11-26:
```
2022-11-26_23_25_43_566: EngineState: confirmation_bluetooth_jdy 115200 alphax 1234:32
2022-11-26_23_25_43_567: EngineState: *** Bluetooth module setup procedure ***
2022-11-26_23_25_44_370: EngineState: Sleeping...
2022-11-26_23_25_45_370: EngineState: Restarting at 115200
2022-11-26_23_25_45_371: EngineState: sending AT  
2022-11-26_23_25_45_371: EngineState: Received 5 +OK  
2022-11-26_23_25_45_372: EngineState: sending AT+VERSION  
2022-11-26_23_25_45_372: EngineState: Received 42 +VERSION=JDY-33-V2.24,Bluetooth V3.0+BLE  
2022-11-26_23_25_45_372: EngineState: sending AT+BAUD  
2022-11-26_23_25_45_373: EngineState: Received 9 +BAUD=8  
2022-11-26_23_25_45_373: EngineState: sending AT+TYPE  
2022-11-26_23_25_45_375: EngineState: Received 9 +TYPE=0  
2022-11-26_23_25_45_375: EngineState: sending AT+PIN  
2022-11-26_23_25_45_378: EngineState: Received 11 +PIN=1234  
2022-11-26_23_25_45_379: EngineState: sending AT+DEFAULT  
2022-11-26_23_25_45_379: EngineState: Received 5 +OK  
2022-11-26_23_25_45_380: EngineState: sending AT+ENLOG0  
2022-11-26_23_25_45_471: EngineState: CRITICAL error: Unexpected SPI device: 2 20221126@microRusEFI
2022-11-26_23_25_45_472: EngineState: Received 5 +OK  
2022-11-26_23_25_45_473: EngineState: sending AT+TYPE0  
2022-11-26_23_25_45_473: EngineState: Received 5 +OK  
2022-11-26_23_25_45_474: EngineState: sending AT+BAUD8  
2022-11-26_23_25_45_571: EngineState: Received 5 +OK  
2022-11-26_23_25_45_571: EngineState: sending AT+NAME=alphax  
2022-11-26_23_25_45_571: EngineState: Received 5 +OK  
2022-11-26_23_25_45_572: EngineState: sending AT+NAMB=alphax  
2022-11-26_23_25_45_671: EngineState: CRITICAL error: Unexpected SPI device: 2 20221126@microRusEFI
2022-11-26_23_25_45_671: EngineState: Received 5 +OK  
2022-11-26_23_25_45_672: EngineState: sending AT+PIN1234  
2022-11-26_23_25_45_672: EngineState: Received 5 +OK  
2022-11-26_23_25_45_673: EngineState: sending AT+RESET  
2022-11-26_23_25_45_673: EngineState: Received 5 +OK  
2022-11-26_23_25_45_674: EngineState: SUCCESS! All commands passed to the Bluetooth module!
```

rusefillc, 2022-11-26:
``microRusEFI``

@dron0gus can you touch the 4chan you have?

dron0gus, 2022-11-26:
@rusefillc I'm tool lazy to solder BT to it. Will do tomorrow.
BT paired and connected to PC (`at+disc` does its work):
```
2022-11-26_23_43_07_910: EngineState: confirmation_bluetooth_jdy 115200 alphax 1234:32
2022-11-26_23_43_07_911: EngineState: *** Bluetooth module setup procedure ***
2022-11-26_23_43_08_328: EngineState: Sleeping...
2022-11-26_23_43_09_363: EngineState: Restarting at 115200
2022-11-26_23_43_09_364: EngineState: sending AT  
2022-11-26_23_43_09_833: EngineState: Timeout waiting for BT reply
2022-11-26_23_43_09_834: EngineState: sending AT+DISC  
2022-11-26_23_43_09_835: EngineState: Received 5 +OK  
2022-11-26_23_43_09_836: EngineState: JDY33 disconnected
2022-11-26_23_43_09_837: EngineState: sending AT+VERSION  
2022-11-26_23_43_10_334: EngineState: Timeout waiting for BT reply
2022-11-26_23_43_10_334: EngineState: sending AT+BAUD  
2022-11-26_23_43_10_335: EngineState: Received 9 +BAUD=8  
2022-11-26_23_43_10_336: EngineState: sending AT+TYPE  
2022-11-26_23_43_10_336: EngineState: Received 9 +TYPE=0  
2022-11-26_23_43_10_337: EngineState: sending AT+PIN  
2022-11-26_23_43_10_337: EngineState: Received 11 +PIN=1234  
2022-11-26_23_43_10_338: EngineState: sending AT+DEFAULT  
2022-11-26_23_43_10_339: EngineState: Received 5 +OK  
2022-11-26_23_43_10_339: EngineState: sending AT+ENLOG0  
2022-11-26_23_43_10_411: EngineState: Received 5 +OK  
2022-11-26_23_43_10_412: EngineState: sending AT+TYPE0  
2022-11-26_23_43_10_512: EngineState: Received 5 +OK  
2022-11-26_23_43_10_512: EngineState: sending AT+BAUD8  
2022-11-26_23_43_10_513: EngineState: Received 5 +OK  
2022-11-26_23_43_10_513: EngineState: sending AT+NAME=alphax  
2022-11-26_23_43_10_610: EngineState: Received 5 +OK  
2022-11-26_23_43_10_611: EngineState: sending AT+NAMB=alphax  
2022-11-26_23_43_10_612: EngineState: Received 5 +OK  
2022-11-26_23_43_10_613: EngineState: sending AT+PIN1234  
2022-11-26_23_43_10_614: EngineState: Received 5 +OK  
2022-11-26_23_43_10_615: EngineState: sending AT+RESET  
2022-11-26_23_43_10_617: EngineState: Received 5 +OK  
2022-11-26_23_43_10_618: EngineState: SUCCESS! All commands passed to the Bluetooth module!
```

dron0gus, 2022-11-27:
AlphaX 4ch with JDY33:
```
2022-11-27_12_01_01_765: EngineState: confirmation_bluetooth_jdy 115200 alphax 1234:32
2022-11-27_12_01_01_765: EngineState: *** Bluetooth module setup procedure ***
2022-11-27_12_01_01_970: EngineState: Sleeping...
2022-11-27_12_01_02_997: EngineState: Restarting at 115200
2022-11-27_12_01_02_998: EngineState: sending AT  
2022-11-27_12_01_03_468: EngineState: Timeout waiting for BT reply
2022-11-27_12_01_03_469: EngineState: sending AT+DISC  
2022-11-27_12_01_03_995: EngineState: Timeout waiting for BT reply
2022-11-27_12_01_03_996: EngineState: Restarting at 9600
2022-11-27_12_01_03_997: EngineState: sending AT  
2022-11-27_12_01_04_469: EngineState: Timeout waiting for BT reply
2022-11-27_12_01_04_470: EngineState: sending AT+DISC  
2022-11-27_12_01_04_569: EngineState: Received 5 +OK  
2022-11-27_12_01_04_570: EngineState: JDY33 disconnected
2022-11-27_12_01_04_570: EngineState: sending AT+VERSION  
2022-11-27_12_01_04_571: EngineState: Received 42 +VERSION=JDY-33-V2.23,Bluetooth V3.0+BLE  
2022-11-27_12_01_04_573: EngineState: sending AT+BAUD  
2022-11-27_12_01_04_669: EngineState: Received 9 +BAUD=4  
2022-11-27_12_01_04_670: EngineState: sending AT+TYPE  
2022-11-27_12_01_04_671: EngineState: Received 9 +TYPE=1  
2022-11-27_12_01_04_672: EngineState: sending AT+PIN  
2022-11-27_12_01_04_674: EngineState: Received 11 +PIN=1234  
2022-11-27_12_01_04_675: EngineState: sending AT+DEFAULT  
2022-11-27_12_01_04_679: EngineState: Received 5 +OK  
2022-11-27_12_01_04_680: EngineState: sending AT+ENLOG0  
2022-11-27_12_01_04_768: EngineState: Received 5 +OK  
2022-11-27_12_01_04_769: EngineState: sending AT+TYPE0  
2022-11-27_12_01_04_869: EngineState: Received 5 +OK  
2022-11-27_12_01_04_870: EngineState: sending AT+BAUD8  
2022-11-27_12_01_04_969: EngineState: Received 5 +OK  
2022-11-27_12_01_04_969: EngineState: sending AT+NAMEalphax  
2022-11-27_12_01_05_069: EngineState: Received 5 +OK  
2022-11-27_12_01_05_069: EngineState: sending AT+NAMBalphax  
2022-11-27_12_01_05_070: EngineState: Received 5 +OK  
2022-11-27_12_01_05_070: EngineState: sending AT+PIN1234  
2022-11-27_12_01_05_169: EngineState: Received 5 +OK  
2022-11-27_12_01_05_170: EngineState: sending AT+RESET  
2022-11-27_12_01_05_171: EngineState: Received 5 +OK  
2022-11-27_12_01_05_171: EngineState: SUCCESS! All commands passed to the Bluetooth module!
```

[image]

rusefillc, 2022-11-27:
4chan with only grey connector
```
2022-11-27_09_37_31_194: EngineState: confirmation_bluetooth_jdy 115200 alphax 1234:32
2022-11-27_09_37_31_194: EngineState: *** Bluetooth module setup procedure ***
2022-11-27_09_37_31_719: EngineState: Sleeping...
2022-11-27_09_37_32_736: EngineState: Restarting at 115200
2022-11-27_09_37_32_736: EngineState: sending AT  
2022-11-27_09_37_32_736: EngineState: Received 5 +OK  
2022-11-27_09_37_32_737: EngineState: sending AT+VERSION  
2022-11-27_09_37_32_739: EngineState: Received 42 +VERSION=JDY-33-V2.23,Bluetooth V3.0+BLE  
2022-11-27_09_37_32_740: EngineState: sending AT+BAUD  
2022-11-27_09_37_32_741: EngineState: Received 9 +BAUD=8  
2022-11-27_09_37_32_743: EngineState: sending AT+TYPE  
2022-11-27_09_37_32_746: EngineState: Received 9 +TYPE=1  
2022-11-27_09_37_32_748: EngineState: sending AT+PIN  
2022-11-27_09_37_32_749: EngineState: Received 11 +PIN=1234  
2022-11-27_09_37_32_749: EngineState: sending AT+DEFAULT  
2022-11-27_09_37_32_750: EngineState: Received 5 +OK  
2022-11-27_09_37_32_750: EngineState: sending AT+ENLOG0  
2022-11-27_09_37_32_802: EngineState: Received 5 +OK  
2022-11-27_09_37_32_802: EngineState: sending AT+TYPE0  
2022-11-27_09_37_32_904: EngineState: Received 5 +OK  
2022-11-27_09_37_32_904: EngineState: sending AT+BAUD8  
2022-11-27_09_37_32_904: EngineState: Received 5 +OK  
2022-11-27_09_37_32_904: EngineState: sending AT+NAMEalphax  
2022-11-27_09_37_33_001: EngineState: Received 5 +OK  
2022-11-27_09_37_33_001: EngineState: sending AT+NAMBalphax  
2022-11-27_09_37_33_102: EngineState: Received 5 +OK  
2022-11-27_09_37_33_102: EngineState: sending AT+PIN1234  
2022-11-27_09_37_33_102: EngineState: Received 5 +OK  
2022-11-27_09_37_33_102: EngineState: sending AT+RESET  
2022-11-27_09_37_33_103: EngineState: Received 5 +OK  
2022-11-27_09_37_33_104: EngineState: SUCCESS! All commands passed to the Bluetooth module!
2022-11-27_09_37_33_312: EngineState: UART: TS: ERROR: no second byte
2022-11-27_09_37_33_312: EngineState: TunerStudio size=748 / total=817 / errors=1 / H=2 / O=355 / P=31 / B=0
2022-11-27_09_37_33_312: EngineState: TunerStudio W=0 / C=0 / P=0

```

rusefillc, 2022-11-27:
I do not know what has changed but it works now. One board was in fact bad soldering, five boards are working good.

I always (?) get false error message

```
2022-11-27_10_58_58_531: EngineState: SUCCESS! All commands passed to the Bluetooth module!
2022-11-27_10_58_58_736: EngineState: UART: TS: ERROR: no second byte
2022-11-27_10_58_58_736: EngineState: TunerStudio size=748 / total=229 / errors=1 / H=2 / O=85 / P=0 / B=0
2022-11-27_10_58_58_736: EngineState: TunerStudio W=0 / C=0 / P=0

```

closed 2022-11-27 by rusefillc

dron0gus, 2022-11-28:
@rusefillc yes, I observe the same. Looks like some glitch on BT module retstart.