rusefi/rusefi #4195: Support individually-wired wasted spark on odd cylinder engines (especially 3cyl)

ElDominio, 2022-05-24:
Add support for individually-wired wasted spark on odd cylinder engines, current example case:

Yamaha FZ09 engine
3cyl
24-2 crank trigger
3 ignition coils

Configured as 3 cylinders fires strangely as expected, had to use as a V6 to get it to work

rusefillc, 2022-05-26:
@ElDominio can you draw a picture of explain how does wasted spark odd cylinders should work?

ElDominio, 2022-05-26:
Think V6 but lop off one bank

rusefillc, 2022-05-26:
[image]

mck1117, 2022-05-26:
Every output fires every 360 degrees. For a 1-2-3 firing order engine, output 1 fires at angle 0, output 2 at 120, and output 3 at 240. Output 1 fires again at 360, then 2 at 480, then 3 at 600.

ElDominio, 2024-02-23:
Does this worl? Mitsubicha runs fine on seuqnetial but have not tested individually wired wasted

ElDominio, 2024-06-15:
@rusefillc does this need testing or something?

rusefillc, 2024-08-30:
related to #6868

rusefillc, 2024-08-31:
problem: original implementation was focusing on engines with even number of cylinders and half the number of coil, like your miata NA/NB. Wasted spark with individual coils was always a hack build on top of that original notion of half-the-cylinders-number-of-coils

in reality individual coils can be used in wasted spark regardless of number of cylinders and even on odd fire two cylinder engines, again as long as induvial coils

one of key parts in the code
```
	// In spin-up cranking mode we don't have full phase sync info yet, so wasted spark mode is better
	// However, only do this on even cylinder count engines: odd cyl count doesn't fire at all
	if (ignitionMode == IM_INDIVIDUAL_COILS && (engineConfiguration->cylindersCount % 2 == 0)) {
		bool missingPhaseInfoForSequential =
			!engine->triggerCentral.triggerState.hasSynchronizedPhase();

		if (!engineConfiguration->oddFireEngine && (engine->rpmCalculator.isSpinningUp() || missingPhaseInfoForSequential)) {
			ignitionMode = IM_WASTED_SPARK;
		}
	}
```

rusefillc, 2024-08-31:
There is now a bounty on this! https://rusefi.com/forum/viewtopic.php?p=49871#p49871

@shadowm60 @nmschulte any chance you guys have time and desire?

shadowm60, 2024-08-31:
I only get near a laptop end of next week

closed 2024-08-31 by rusefillc