rusefi/rusefi #4228: TPS error usability

rusefillc, 2022-06-02:
https://rusefi.com/online/view.php?msq=1045
https://rusefi.com/online/view.php?log=854

1. https://files.slack.com/files-pri/T4NJHQ8QZ-F03JAS6GSHF/image.png is this kosher? if it's not kosher we need a red warning to appear right on this dialog

2.
```
[Jimmy AssTune Luntzel](https://app.slack.com/team/U010VNQSZM0)  [10:46 PM]
"Confirmed tps 1 and 2 reading correctly"
"There is tps error"

Confirmed by moving throttle by hand and seeing both sensor values read 5-0v and 0-5v
That is what I confirmed
"tps error" does not tell me what the error is
```

duhafnusa4, 2022-06-02:
[image]

duhafnusa4, 2022-06-02:
[image]

[image]

[image]

duhafnusa4, 2022-06-02:
[image]

mck1117, 2022-06-02:
@abelom went down the wrong path of troubleshooting, not listening to the user when he said "the auto calibrate button does nothing". The TPS works fine, but was just uncalibrated because the throttle wouldn't move (for some other reason, TBD)

rusefillc, 2022-06-02:
@mck1117 it looks like we need a more obvious way to handle "the auto calibrate button does nothing" so that user is clearly focused on that being his main problem

mck1117, 2022-06-02:
We were so close.

[image]