rusefi/rusefi #4243: Cheap password protection

ElDominio, 2022-06-06:
I have many requests of tuners to allow the use of password protection. It doesn't matter to me if the maps themselves are encrypted or not (would complicate things much more)

Only be able to disallow reading/writing through serial until the password is entered, preferably in a submenu under controllers, at which point the ECU correctly responds to read/write requests from Tunerstudio.

Once the correct password is entered, the  ECU entered an "unlocked mode", where the tune can be read/written until the lock is reapplied. Could be done with a dialog box with a password entry and a checkbox, once the box is unchecked, the ECU throws an error to ask the user to reconnect for map access.

rusefillc, 2022-06-06:
good solution is not needed, we will start with a lame solution:

1) new flag which would disable write commands
2) padlock buttons to unlock
3) TBD mechanism for password entry and persistance

NMSTEC, 2022-06-07:
I like this solution.

ElDominio, 2022-07-22:
I was asked to ping two weeks after the ticket was opened so here id a ping a month afterwards

rusefillc, 2022-07-25:
Sounds like https://rusefi.com/forum/viewtopic.php?p=34942#p34942 is the way to go here :(

I might have a cadidate for paid development assignments but I do not see myself personally coding this feature in the coming months.

closed 2022-07-30 by rusefillc

rusefillc, 2022-07-30:
[image]

ElDominio, 2024-10-14:
Question: to unlock, can i just blast "unlock [password]" through serial? 

I want to make a neat unlock app so people can use password without consome

rusefillc, 2024-10-15:
@ElDominio no, you would wrap the command into a packet with tunerstudio protocol header and footer including CRC

ElDominio, 2025-05-26:
> [@ElDominio](https://github.com/ElDominio) no, you would wrap the command into a packet with tunerstudio protocol header and footer including CRC

Even for unlocking? Not locking?

commit 26ae6f74 (2022-07-30): [SECURITY] Cheap password protection against tune modification fix #4243
commit 945e8766 (2022-07-30): [SECURITY] Cheap password protection against tune modification fix #4243
commit e0677fda (2022-07-30): [SECURITY] Cheap password protection against tune modification fix #4243
commit c81ede01 (2022-07-30): [SECURITY] Cheap password protection against tune modification fix #4243

helping build