rusefi/rusefi #4248: Hellen Miata ETB pins

BeerMoneyMotorsports, 2022-06-10:
Please unhide the Hellen Miata ETB pins selection.  This is for all 4 Miata Hellens.

[image]

rusefillc, 2022-06-10:
@BeerMoneyMotorsports based on your YAML PRs you have the skills to PR this. Please give it a try?

That would help reduce some of my load. Let me know if this is urgent and you need help ASAP

BeerMoneyMotorsports, 2022-06-10:
ill try to figure it out.  Per @mck1117 recommendation, I will hard code them.

ETB_EN - C6
+ETB - C7
-ETB - C8

BeerMoneyMotorsports, 2022-06-10:
[image]