rusefi/rusefi #4251: Sense resisters for injectors should be 15mOhm, or settings altered

NOx-z, 2022-06-10:
all of these sense resistors for Rev A are 10mOhm whereas all of our previous versions are 15mohm (except for 10mOhm used on the boost converter sense, sense number 4). This has implications of incorrectly setting the expected peak-hold currents with our current settings.

solutions:
A) change settings for 10mOhm
b) swap resistors to 15mOhm

rusefillc, 2022-06-10:
I will fix my legacy board, who else has what legacy boards? @NOx-z @mck1117 do you have any legacy? safer to update those all

rusefillc, 2022-06-10:
Does this sound like https://github.com/rusefi/rusefi-hardware/issues/92 reason or contributing factor?

rusefillc, 2022-06-10:
is this related to https://github.com/rusefi/rusefi-hardware/issues/96 at all?

mck1117, 2022-06-11:
@rusefillc can you confirm what resistors are on the board you started the car on previously? There are references in code to them being 10mOhm, but we thought they were maybe 15.

rusefillc, 2022-06-16:
Flight delayed by 40 minutes, ETA touch down 11pm EST

rusefillc, 2022-06-17:
So "R22,R12,R26,R21"?

https://rusefi.com/docs/ibom/Common_Rail_MC33816_0.3.2.html ?

i think https://rusefi.com/forum/viewtopic.php?p=37790#p37790 is the picture of my actual board

at least  R010 matches 

[image]

mck1117, 2022-06-17:
OK, in that case we're all set here, no change to make.

closed 2022-06-17 by mck1117