rusefi/rusefi #4252: junk dreams: 6 pin GM ETB connector

rusefillc, 2022-06-12:
looks like early SENT ETBs were using 6 pin plug with one pin unused

duhafnusa4, 2022-06-12:
Add calebchesh to the junkyard stuff

NMSTEC, 2022-06-13:
Autozone will have this.

rusefillc, 2022-06-13:
@NMSTEC please help with best keywords or part number or link?

NMSTEC, 2022-06-13:
[image]
[image]

NMSTEC, 2022-06-13:
https://www.amazon.ca/ACDelco-PT2713-Original-Equipment-Multi-Purpose/dp/B007QC2X0A

rusefillc, 2022-06-13:
i need the flat six pins

[image]

rusefillc, 2022-06-13:
[image]

rusefillc, 2022-06-13:
@NMSTEC see ^^^ I need the flat style closer to the 5 pin flat on the ETB you gave me

actually funny thing: the wrong pigtail you've shipped to me is very similar - it is 6 pin and it is flat - but it's 20% too large. i.e. there are multiple sizes in that style :(