rusefi/rusefi #4255: Ability to have WBOs heated upon ignition

NMSTEC, 2022-06-17:
I am well aware of dangers that are there for preheating o2s with engine off, but in this particular case I need the O2s heated upon ignition on. 

Either warning, or some sort of bypass. Or if I can cheat with LUA. This is temporary and will be shut off once vehicle is finished.

NMSTEC, 2022-06-21:
Bump. Need this please.

dron0gus, 2022-06-21:
If there was no CAN WB_MGS_ECU_STATUS message received since last power on and battery voltage is above 9.5V for 3 seconds it will start heating.
See https://github.com/mck1117/wideband/commit/bc0686c82671a07e49078f76af0066841277f759#diff-0ae7685ffdbc02debe8fdffbcbd5cabeb7c6b075f827ea343e73549bc2979bf7

mck1117, 2022-06-21:
@dron0gus the onboard wbo controllers on hellen have no battery sense, so that's not true

BeerMoneyMotorsports, 2022-06-22:
Apply a true/false option for heating before startup. TS note “heating before cranking leads to short sensor life. Use at own risk!”

PR #4275 (2022-06-23): add option to force o2 heating
fix #4255