rusefi/rusefi #4262: rev limiter real test data

rusefillc, 2022-06-20:
we have multiple means to limit revs

we have very hard rev limited on the global dialog

do we consider that rev limiter strategy good, great or not so?

We need a reference log and video of test mule hitting that hard limit at 3000

NMSTEC, 2022-06-20:
Will supply tomorrow with both.
Would like to note, soft limit is beautiful.
https://rusefi.com/online/view.php?log=906
Soft set to 6800 with hysteresis 100rpm

rusefillc, 2022-06-20:
i think I forgot about tune - let's make it a combo of tune+log+video focusing on just specific use-case

i will package that info for https://github.com/rusefi/rusefi/wiki/Case-Studies

NMSTEC, 2022-06-20:
I cannot recall which specific variation of tune I used for this log. Will make new multiple sets tomorrow! Would you like 3k rpm specific? Or multiple rpms?

rusefillc, 2022-06-20:
``We need a reference log and video of test mule hitting that hard limit at 3000``

OrchardPerformance, 2022-06-20:
I can give you logs from the drift car for both normal hard cut and soft+hard limit.

But XY problem. What are you hoping to see in the logs? 

Drift car was breaking pistons so I looked a lot at the hard cut behavior, currently I would say it is "fine", I.e. no better or worse than the competition. 

If we were to look at improvements I would suggest:

"Ignition retard soft limiter" - like the function of the etb limiter, reduce power near rev limit by retarding spark to soften the approach to the rev limit.
Simple option of "degrees retard at limiter" + "rpm to start retarding", simply interpolate from that start rpm to chosen limit rpm.

Add an option for "retard after hard cut", this would be an option to offer a number of degrees the timing would be retarded by after a hard cut event, this would soften the re-application of power after a hard cut and would prevent lean detonation events near the limiter. 
Table for re-advance rate would be needed, exepct the best way is to have a table like the cranking duration multiplier but using degrees retard vs cycles since hard cut triggered.

Finally we should probably take account of wall wetting fuel.
After a fuel cut the port walls will be dry, the next injection will wet the port walls and be lean, potentially all first injections after a hard fuel cut should be the size of the first cranking pulse?
This is an open question as I do not know the best way to improve it.

rusefillc, 2022-06-20:
@OrchardPerformance i had to remove misplaced improvements suggestsion. i know you have invested time into typing but I need to invest in separation of issues

this ticket is requesting a combination of tune, log and video. a subset of the dataset would have negative value by not giving my overall picture of how different kinds of RPM limit look, sound and smell

BeerMoneyMotorsports, 2022-06-20:
@rusefillc Will free rev work or do you prefer while under load/driving?

rusefillc, 2022-06-20:
@BeerMoneyMotorsports free rev to begin with for the most reproducible experiment

just comparing three sets of apples to each other for now (hard, soft via ETB, skipped spark via Lua)

rusefillc, 2022-06-21:
@NMSTEC can we have same of hard limit?

```
For example, we have very hard rev limited on the global dialog. 
```

@mck1117 @OrchardPerformance opinions on why above "soft at 3000" looks so hard on the videos?

NMSTEC, 2022-06-22:
I would not say thats hard. When HARD kicks in, theres actual flames out the exhaust, and very violent noises. Soft just feels like im and on off the throttle. Contributing factors are size of engine, and intake. This thing picks up speed VERY quickly, and just overshoots its soft limit. And having hysteresis of 150 probably isnt any use. I'd presume 500 rpm hysteresis would be way smoother. 

Once I reassemble E67, I will do hard cut log and video.

mck1117, 2022-06-22:
@rusefillc have a look at the log - it's pretty darn smooth. AFR is well controlled the whole time. It'd the same as just mashing the pedal.

OrchardPerformance, 2022-06-22:
It only looks hard because energy balance, soft limiter will not be able to stop that much power over speeding the crank before it can kick in. 

This is a topic that will get really fucking complex really fast. Doing this here is likely to not work well. 

Question -is it worth taking the time to deep dive on forum then answer question on the phone? 
- Tuning Support thread?