rusefi/rusefi #4265: Add EGT support or remove auxtemp temp. limit

elmepro, 2022-06-21:
Can we ad EGT temperature support, or remove (auxtemp1/2) temperature limitation?
[image]

elmepro, 2022-06-21:
Look like i was no aware that EGT requires module.

Im using: [https://www.aliexpress.com/item/4000924596475.html?spm=a2g0o.order_list.0.0.26f81802szBPky](url)

And this should be the module (MAX31855) : [https://www.aliexpress.com/item/32878757344.html?spm=a2g0o.productlist.0.0.42803ffa1e82mx&algo_pvid=5392958e-5000-4068-9ed8-ced1f54e126b&algo_exp_id=5392958e-5000-4068-9ed8-ced1f54e126b-7&pdp_ext_f=%7B%22sku_id%22%3A%2265723389010%22%7D&pdp_npi=2%40dis%21EUR%21%214.19%21%21%21%21%21%4021031a5516558424199065978e024f%2165723389010%21sea](url)

mck1117, 2022-06-21:
what ECU? how is the max31855 wired?

closed 2022-06-21 by rusefillc

TurboMarian, 2023-01-08:
Is EGT implementation done already? I cannot connect MAX31855 via SPI to STM32F407VGT6 dev board. Everytime I'm trying to select SPI 1 or SPI 2 communtication with TS is broken after restart, Both SPIs are working fine with other devices like SDCARD...

rusefillc, 2023-01-08:
@TurboMarian we have MAX31855 and I am sure that it would work if configuration is legit