rusefi/rusefi #4266: Hellen Miata NA 96/97 Interactive pinout points to wrong page

stefanst, 2022-06-22:
The link in the wiki here:
https://github.com/rusefi/rusefi/wiki/Hellen-NA96
points to the interactive pinout for the Miata NB1

rusefillc, 2022-06-22:
https://github.com/rusefi/rusefi/wiki/Hellen-NA96 points at https://rusefi.com/docs/pinouts/hellen/hellenNA8_96/ now

https://github.com/rusefi/rusefi/blob/master/firmware/config/boards/hellen/hellenNA8_96/connectors/main.yaml says " Hellen Miata NB1" does not seem right

https://github.com/rusefi/rusefi/blob/master/firmware/config/boards/hellen/hellenNA8_96/connectors/main.yaml mentions main.jpg but no such file in https://github.com/rusefi/rusefi/tree/master/firmware/config/boards/hellen/hellenNA8_96/connectors folder

rusefillc, 2022-06-22:
I've at least copy-pasted main.jpg file

rusefillc, 2022-06-22:
@BeerMoneyMotorsports please take care of yaml fixes at least title maybe more

reminder that all pinout pages can have back-link like top of https://rusefi.com/docs/pinouts/proteus/ does

closed 2024-12-27 by rusefillc