rusefi/rusefi #4284: Smarter return-to-idle logic

mck1117, 2022-06-24:
Rate-of-return/deltaRPM based enable for idle logic, i.e. RPMs falling faster than X/second

If RPM is headed toward idle quickly, enter idle before we actually get there. Can help avoid stalling before the engine overshoots the target and stalls.

ElDominio, 2022-06-27:
Very related to https://github.com/rusefi/rusefi/issues/3940 I think, as soon as this is implemented I will test

NMSTEC, 2022-06-27:
I too can and will test this. Our stunt vehicles return to idle awfully fast when stunt brake is applied.

ElDominio, 2022-07-16:
I think this, #3940  and #4283 should all be rolled up into one. Coast -> idle should have a transition period to help coming down to idle, that basically solves the problems of these three issues

ElDominio, 2022-08-21:
Is there any way idle duty control can be added to Lua? I'll implement a rough sketch that should be good enough until this is done

rusefillc, 2022-08-21:
Nope
#4465

ElDominio, 2022-08-30:
I recorded a video of how LinkECU works in terms of this

In this video, base idle is 2.3%, and dashpot adder is 0.5%, idle Hold time is 0.5s, and decay time is 0.5s

I screwed up a bit on the graph scaling but you can see it all happening by the numbers, and it also shows RPM and how it behaves when this function is working.

https://user-images.githubusercontent.com/5685108/187349628-5951a4e4-df90-408f-b232-c132f3eac1ce.mp4

ElDominio, 2023-03-21:
I just thought of maybe an easier way to think about this

The moment the ECU would turn on the green "Idling" bulb in tunerstudio, a 2% offset in IAC is added, then slowly removed over time

Base condition: 25% IAC 
car is idling
car goes into Coasting/Cruising
+2% is added to IAC
IAC is now 27% in coasting
car comes down from Coasting
state changes to Idling (TPS under threshold, TunerStudio bulb is lit green)
countdown starts, for .5 seconds we will hold IAC at 27%
after .5s have passed, we want to bring down IAC back to 25%, so 
we have a transition time (for this example .5s) that we will use to gradually lower IAC down to 25%
1s has passed after the "Idling" bulb has come on, we are now back to 25% IAC
idle is happy

I am trying to write in Lua but timer function is complicated