rusefi/rusefi #4292: Time delay before DFCO

ElDominio, 2022-06-27:
I remember some technical reason for this behavior but I don't remember it. The real  reason i would like this is so cars with pops can have, for example, 2-3 seconds of pops then go into DFCO. 

I think the technical reason is to control the amount of engine braking available but I'm not going to lie lol

ElDominio, 2022-08-16:
Any progress on this?

10AE0739, 2022-08-24:
I was also looking for this feature in settings.

rusefillc, 2022-08-26:
@mck1117 shall we close this ticket?

mck1117, 2022-08-26:
@ElDominio want to give it a try?

ElDominio, 2022-08-26:
Of course, I'll try to get it tested before sunday

10AE0739, 2022-08-31:
I'm running the 8/30 snapshot on my Hellen NB1. This works well. Thank you.

mck1117, 2022-08-31:
Reports of it working, looks like we're done here!

closed 2022-08-31 by mck1117

rusefillc, 2022-08-31:
We have this one and #4500 I wonder what else can use a time delay