rusefi/rusefi #4293: Be able to use CAN RPM/TPS/MAP/VSS as operating inputs (replace actual sensors)

ElDominio, 2022-06-27:
I have been wondering if this is possible, as this could be useful to use 2Chan as aux fuel/boost control/additional I/O for cars that need this (not necessarily running rusEFI)

In my case I have a Hyundai Kona 1.6T in which 2Chan could be controlling aux fuel (only need injection time to be correct, since these are not the main injectors), boost control, and probably with flex fuel input.

mck1117, 2022-06-27:
You can already set sensors from Lua (and receive CAN, of course).  That works for everything other than RPM, but you could do it by just PWMing the injectors without "running" the engine.

ElDominio, 2022-06-27:
but, boost control is RPM dependent too, no? And for additional I/O conditions, most of those I/O would be RPM dependent

rusefillc, 2022-06-27:
use Lua with one of lua general purposes tables not boost control table?

rusefillc, 2022-06-27:
> you could do it by just PWMing the injectors without "running" the engine.

i do not get this part

rusefillc, 2022-06-28:
@mck1117 ^^^

rusefillc, 2022-07-22:
https://github.com/rusefi/rusefi/wiki/Lua-Scripting#set-sensor-value has an example

ElDominio, 2022-07-22:
Can it set RPM? and get treated as trigger? (Could be an evently spaced pulse at x frequency divided by number of cylinders?)

rusefillc, 2022-07-22:
@ElDominio nope, RPM would be a very, very separate story

ElDominio, 2023-07-23:
I think this works now right?

rusefillc, 2023-07-23:
Yes possible with lua

closed 2023-07-23 by rusefillc