rusefi/rusefi #4294: Timed boost drop after input or condition

ElDominio, 2022-06-27:
This is useful for automatic transmission cars where between gears boost can spike. A triggerable boost reduction that is held for an amount of time, or until RPM drops by a certain amount, to assist poor weak transmissions during shifting

rusefillc, 2022-06-27:
please list all configurable settings

what would be triggering the boost drop?

mck1117, 2022-06-27:
Would lua hooks for boost control complete this?

NMSTEC, 2022-06-27:
Is the usual way not timing adjusted for torque reduction?

ElDominio, 2024-02-23:
I did it with Lua as @mck1117  suggested. 

@NMSTEC , removing timing raises boost since load is increased.

rusefillc, 2024-02-23:
@ElDominio new label created "need-to-trickle-down"

please provide your Lua script?