rusefi/rusefi #4295: usability confustion: ediatble fields with board overrides

rusefillc, 2022-06-27:
We have hard-coded board overrides and for instance in case of VBAT we have field editable

We shall make this less confusing, I assume TS has options for read-only values?

rusefillc, 2022-06-27:
https://user-images.githubusercontent.com/48498823/175981287-2c19d393-b4e5-49ff-b6b5-85cf66367312.mp4

mck1117, 2022-06-27:
just hide them on boards we know override it

mck1117, 2022-06-27:
aka, the user doesn't ever have to think about battery sense on a proteus/mre/hellen/whatever, so why even show the UI?

rusefillc, 2022-06-27:
where do we draw the line?

the value of showign read-only value would be sharing technical details while also sending the message that it's hard coded

like for instance CLT pull-up should it be hidden like I am about to push?

commit b231b458 (2022-06-27): usability confustion: ediatble fields with board overrides fix #4295
closed 2022-06-27 by rusefillc