rusefi/rusefi #4297: [SECURITY] hiding tune

rusefillc, 2022-06-28:
We would like a way to make all or portion of the tune, for instance Lua script, absolutely secret from external world

rusefillc, 2022-06-28:
Question: would "read protection level 1" combined with change to READ_CHUNK protocol command implement the "hide tune which is already inside ECU" task?

See also
https://www.st.com/resource/en/application_note/an4701-proprietary-code-readout-protection-on-microcontrollers-of-the-stm32f4-series-stmicroelectronics.pdf

rusefillc, 2022-06-28:
@mck1117 says "Stm32 debug protection bits are broken pretty easily with power supply glitching"

mmatousek89, 2022-07-16:
im +1 on some sort of security capabilities of ecu :)

rusefillc, 2023-10-18:
https://community.st.com/t5/stm32-mcus/how-to-change-the-read-out-protection-on-stm32f4/ta-p/49410