rusefi/rusefi #4302: reduce generated .h in git

rusefillc, 2022-06-29:
At least some generated headers should not be checked into git at all

in the meantime proper single line make should be happy meaning that makefile should be managing lazy generation of at least some of the headers?

rusefillc, 2022-08-17:
At this point I am convinced that this ticker here is more important or at least equally imporant as newparse.

@mck1117 @chuckwagoncomputing any chance anyone would be available to help?

rusefillc, 2022-08-17:
probably not related to #4454

chuckwagoncomputing, 2023-03-02:
Is there any reason that Make couldn't handle config generation?

rusefillc, 2023-03-02:
no reason I am aware of, just a question of identifying next step

chuckwagoncomputing, 2026-01-07:
## Benefits to removing generated files

- CI pushing to main has always been problematic; it seems to be working now, but there's also reduced contribution volume right now
- Reduced repo size (very rough estimate ~200MB)
- Simpler workflow for contributors who don't have much git experience
- It's just weird to be storing build artifacts in the git repo
- Having inis in the repo could be a source of confusion if new users try to use them with a release build - not sure if this has ever happened

chuckwagoncomputing, 2026-01-13:
Waiting on #8854