rusefi/rusefi #4310: EMI kills USB and causes a FATAL on the ECU

rusefillc, 2022-07-01:
we at Tutomo

laptop sits on top 3 phase fan box connected to a anderson VFD

ECU is happy 6 feet away
connect UBS
ECU is red FATAL
USB does not recognise ECU

rusefillc, 2022-07-01:
Tutomo is a JDM show with hub dyno in Newark, NJ friend of @blundar 

At the moment tuning hellen88bmw

rusefillc, 2022-08-31:
Not actionable without steps to reproduce.

Next step: reproducing this on a bench, need some source of brutal EMI.

@mck1117 @andreika-git @stefanst any ideas?

mck1117, 2022-08-31:
No idea. I've never seen this on a Proteus. Is this a 2 or 4 layer board?

rusefillc, 2022-08-31:
https://github.com/rusefi/hellen88bmw/blob/main/boards/readme.md says 4 layers

rusefillc, 2023-10-20:
Same VFD got https://rusefi.com/s/scan to go red

* with just one piece USB-C cable I was just getting USB connection dropped
* Tommy has brought his favorite USB extension cable and with USB extension we got red FATAL LED - unit has +12v coming from vehicle ODB

next I should plan a trip with a collection of different ECUs and debugger

[image]

dron0gus, 2023-10-20:
https://aliexpress.ru/item/1005001369085297.html?sku_id=12000029660837645&spm=a2g2w.productlist.search_results.0.36944aa6AV7t2e&gatewayAdapt=glo2rus

rusefillc, 2023-10-21:
small, not-debug-build

only happens if small is powered by +12 from unhooked vehicle, only happens when laptop is hooked to A/C charger

```
2023-10-21_15_33_19_897: ConsoleUI: COMPOSITE_OFF_RPM=700
2023-10-21_15_33_21_044: EngineState: Last boot had hard fault type: 3 addr: 20020010 CSFR: 82
2023-10-21_15_33_21_047: EngineState: r0  0x8060984
2023-10-21_15_33_21_048: EngineState: r1  0x1
2023-10-21_15_33_21_048: EngineState: r2  0x8060984
2023-10-21_15_33_21_048: EngineState: r3  0x1
2023-10-21_15_33_21_050: EngineState: r12 0x1
2023-10-21_15_33_21_050: EngineState: lr (thread)  0x80073E7
2023-10-21_15_33_21_052: EngineState: pc  0x80066CC
2023-10-21_15_33_21_053: EngineState: xpsr  0x1000053
2023-10-21_15_33_21_054: EngineState: Power cycle count: 1
2023-10-21_15_33_21_055: EngineState: initHardware()
```

rusefillc, 2023-10-21:
debug build of small-can
https://rusefi.com/build_server/rusefi_bundle_small-can-board_debug.zip

```
2023-10-21_15_43_53_704: EngineState: Last boot had hard fault type: 3 addr: 20020010 CSFR: 82
2023-10-21_15_43_53_707: EngineState: r0  0x807B1DC
2023-10-21_15_43_53_710: EngineState: r1  0x807B42C
2023-10-21_15_43_53_713: EngineState: r2  0xC9
2023-10-21_15_43_53_715: EngineState: r3  0x807B1DC
2023-10-21_15_43_53_717: EngineState: r12 0xF4240000
2023-10-21_15_43_53_719: EngineState: lr (thread)  0x8006FB5
2023-10-21_15_43_53_721: EngineState: pc  0x8046AFA
2023-10-21_15_43_53_722: EngineState: xpsr  0x21000053
2023-10-21_15_43_53_724: EngineState: Power cycle count: 1
```

rusefillc, 2023-10-21:
```
2023-10-21_16_10_59_559: ConsoleUI: COMPOSITE_OFF_RPM=700
2023-10-21_16_11_00_135: EngineState: Last boot had error:    0 M  q  &     5Q    ; g3  E  hsV 6 @   V4     b  =y *  Q*      R       G      C  b  2Y A        @ 
2023-10-21_16_11_00_136: EngineState: Last boot had hard fault type: 3 addr: 20020010 CSFR: 82
2023-10-21_16_11_00_137: EngineState: r0  0x807B1EC
2023-10-21_16_11_00_138: EngineState: r1  0x807B43C
2023-10-21_16_11_00_138: EngineState: r2  0xC9
2023-10-21_16_11_00_139: EngineState: r3  0x807B1EC
2023-10-21_16_11_00_139: EngineState: r12 0xF4240000
2023-10-21_16_11_00_140: EngineState: lr (thread)  0x8006FB5
2023-10-21_16_11_00_141: EngineState: pc  0x8046B0A
2023-10-21_16_11_00_142: EngineState: xpsr  0x21000053
```

rusefillc, 2023-10-21:
CH_CFG_SYSTEM_HALT_HOOK invokes chDbgPanic3 which invokes criticalError which invokes firmwareError 

```
2023-10-21_16_23_13_374: EngineState: Last boot had raw:        qz?  ( = j  W(  ;j ?1F0 1     K V9   $  -  9    b           e1   m   ]8     M 5&-m [  u  Y 8 5 
2023-10-21_16_23_13_376: EngineState: Last boot had hardFile:     <    
2023-10-21_16_23_13_378: EngineState: Last boot had line: -559038737
2023-10-21_16_23_13_379: EngineState: Last boot had error:    0 M  q  &     5Q    ; g3  E  hsV 6 @   V4     b  =y *  Q*      R       G      C  b  2Y A        @ 
2023-10-21_16_23_13_381: EngineState: Last boot had hard fault type: 3 addr: 20020010 CSFR: 82
2023-10-21_16_23_13_382: EngineState: r0  0x807B26C
2023-10-21_16_23_13_383: EngineState: r1  0x807B4BC
2023-10-21_16_23_13_384: EngineState: r2  0xC9
2023-10-21_16_23_13_385: EngineState: r3  0x807B26C
2023-10-21_16_23_13_386: EngineState: r12 0xF4240000
2023-10-21_16_23_13_387: EngineState: lr (thread)  0x8006FB5
2023-10-21_16_23_13_387: EngineState: pc  0x8046B3E
2023-10-21_16_23_13_388: EngineState: xpsr  0x21000053
2023-10-21_16_23_13_389: EngineState: Power cycle count: 1

```

rusefillc, 2023-10-22:
@Qwerty-OFF says bring oscilloscope
@Qwerty-OFF says bring capacitors for power line

rusefillc, 2023-12-17:
still reproduces:

* v20231216 debug build
* power from white civic
* laptop on A/C charger on top of VFD

rusefillc, 2023-12-17:
as is power line

[image]

rusefillc, 2023-12-17:
[image]

[image]

rusefillc, 2023-12-17:
[image]

[image]