rusefi/rusefi #4311: Ignition timing drift correction

rusefillc, 2022-07-01:
Tommy says that on some vehicles you need a configurable by rpm time offset if timing to get say 10  degrees fixed timing on the whole RPM range

Something something cam only sensors, belt stretch etc

mck1117, 2022-07-01:
I'd love to add it when shown a video of a car set to "fixed timing" mode and a timing light that shows drift

rusefillc, 2022-07-01:
Sounds like a plan. Closing for now, search would remember.

closed 2022-07-01 by rusefillc

rusefillc, 2022-07-02:
@eldomonio @blundar opinions?

ElDominio, 2022-07-02:
I mean, as I usually say, most other ECU's have this. I have never really tested this happening and when you do the tune on the dyno, you usually end up compensating for this in the ignition map anyways.

But it is usually an option on pretty much every other ECU.
[image]