rusefi/rusefi #4313: re drivers not installing

BeerMoneyMotorsports, 2022-07-02:
Hardware- Hellen Miata all
Firmware- current Release and Snapshot.  This has been an issue for a little while however i do not know the full scope.  Sometimes "install drivers works fine, other times it doesn't.

For new users that need to update firmware, they are following the instructions to
- Download correct firmware
- Connect ECU to laptop
- Open Console
- click "Install Drivers"
- restart laptop
- open console and hit Auto DFU
- "ERROR- DEVICE NOT INSTALLED"
- In device manager you see STM32 Bootloader with the caution triangle

The fix- 
- open device manager while ecu still in DFU
- update driver
- point it to the file folder ''Drivers" in the firmware.  must point it to the entire folder and not the individual driver
- install
- cycle ecu power and open console
- auto DFU now works.

BeerMoneyMotorsports, 2022-07-02:
[image]

rusefillc, 2022-08-23:
Got another complaint

We need to test if driver installer works for STM32 bootloader

Minor progress proposal #4475

As an idea shall we display any sort of helpful URL?

[image]

rusefillc, 2022-08-24:
See also #4476

closed 2022-08-24 by rusefillc

rusefillc, 2022-08-28:
Do we fail to close one of the windows or something else with one of the windows?

[image]

commit 67b4ccc7 (2022-08-24): STM32Bootloader driver install from console requests elevation fix #4313