rusefi/rusefi #4314: Dyno CAN bus integration

rusefillc, 2022-07-03:
Talking to the dyno operator (Tommy) yesterday, he explained how some aftermarket ECUs talk to the dyno using CAN bus communication which enables automating some tasks of the tuning game. He mentioned changing VE cells. Of course I'm thinking at this point, especially if you have ETB, you could technically fully automate the tuning process. The dyno takes control, sets the rpms to the first rpm cell. The ECU goes through the throttle positions to hit the points on the load axis. Reading the WB, we automatically adjust VE. Move on to the next rpm cell. Do the same for the ignition table, finding MBT. And voila, you have the two most important tables tuned.

rusefillc, 2022-07-03:
@ElDominio @blundar @NMSTEC do you know anything about this? what dynos are integrated with what ECUs? what kind of features?

ElDominio, 2022-07-03:
Mainline and Dynocom have this

On the ECU side, Haltech and MoTeC afaik have this and yep, it's as you say. Usually you need to put the dyno in a certain mode which sends a bit to the ECU that it's time for the comms to start. Also, i THINK either on dyno or ECU side some limits are established for safety's sake

rusefillc, 2022-07-03:
That's dynojet shit according to @NMSTEC

@stefanst would you be available to email Mainline, Dynocom and dynojet and inquire on CAN documentation?

NMSTEC, 2022-07-03:
Afaik that only applies to dynojets power commander for motorcycles.

stefanst, 2022-07-03:
I'll see what I can find out.
One more data point: The dyno manufacturers charge the users for unlocking SW features. As per dyno-owner Tommy, this is per ECU brand. We may want to piggyback onto another brand's protocol if possible/legal to reduce cost for the dyno owner/operator and to make it easier to implement on multiple platforms.

stefanst, 2022-07-03:
I contacted mainline and dynocom.

pkagiantas, 2022-07-03:
yourdyno has it too you might be able to work with Jostein Fonneland ( on facebook ) to make it happen 
and i have a yourduno daq on a rolling road if you ever need anything :)

rusefillc, 2022-07-03:
@pkagiantas thank you! i've messaged Jostein Fonneland on FB please tell him he has that request to make sure he notices :)

rusefillc, 2022-07-03:
spoke to Jostein 

phase 1: of the implementation: ECU sends to dyno, dyno decodes for logging. he has one way from ECU to dyno in yourdyno 

phase 2: dyno actively controlling ECU, for instance to sweep ignition range while at fixed RPM. yourdyno does not have dyno controlling ECU at the moment

open question if any of the dyno just reads .DBC, yourdyno does not. yourdyno  is not open source.

rusefillc, 2022-07-03:
https://forums.haltech.com/viewtopic.php?t=34730#p72694 has Haltech-CAN-Broadcast-Protocol-V2.35.0.pdf attached

[Haltech-CAN-Broadcast-Protocol-V2.35.0.pdf](https://github.com/rusefi/rusefi/files/9035514/Haltech-CAN-Broadcast-Protocol-V2.35.0.pdf)

stefanst, 2022-07-04:
Hi Stefan,

I was forwarded you email requesting data integration between your rusEFI and DynoLog, from Mainline.

We are more than happy to add support for your ECUs. Do you have an existing CAN-Bus protocol, eg, for
sending to Dashes? If you have your own protocol, please send me the details and I'll look at adding it
to DynoLog.

If you don't have an existing protocol, I can send you a generic one that we have devised for this purpose.

-- 
Regards,
 
Andrew Elder
DynoLog Dynamometer Pty Ltd
PO Box 151, Montrose, 3765
Victoria, Australia
Ph +61 0408 055 541
Email [aelder@melbpc.org.au](mailto:aelder@melbpc.org.au)

rusefillc, 2022-07-04:
While we have https://github.com/rusefi/rusefi/blob/master/firmware/controllers/can/rusEFI_CAN_verbose.dbc I wonder if imlementing the generic one on our side is the quicker and safer way forward

stefanst, 2022-07-04:
That's above my paygrade.

rusefillc, 2022-07-04:
@stefanst can you please CC me on the email chain?

NMSTEC, 2022-07-04:
Spoke to a few tuners locally, (One who uses it for a living tuning motorcycles) says this system is amazing for NA engines. For turbo engines it is not. It takes so long to go through cells in boost that the whole manifold and turbo start glowing. With the IATs going through the roof, that does not usually get accounted for in VE or timing.

rusefillc, 2022-07-04:
We have Mainline format at #4319

this ticket is now focusing on dynos other than Mainline

rusefillc, 2022-08-13:
See also #4319 for the other direction

rusefillc, 2024-09-21:
Most useful feature:

dyno sends, ECU receives HP and TQ
a button to start dyno run

rusefillc, 2024-09-21:
IL has dyno dynamics dtm500