rusefi/rusefi #4315: Extend the posibility of using analog input as digital

shadowm60, 2022-07-03:
Situation:
On hellen128 only C40 is of type din.
This way only this pin can be used for starter button, launch control input.

Need to find a way to extend this and also include there the unused analog inputs but without loosing the possibility of also use those as analog in case no need for digital.

mck1117, 2022-07-03:
just add them as switch inputs in the yaml, it works great

shadowm60, 2022-07-03:
- pin: P40
    # H144_IN_MAP3 AIN15 A15 PA2
    id: EFI_ADC_2
    class: analog_inputs
    ts_name: P40_IN_MAP3
    type: [av, din]

shadowm60, 2022-07-03:
would this work ok?

rusefillc, 2022-07-03:
@shadowm60 https://github.com/rusefi/rusefi/commit/752a8af70a3985d2f9692695cb31384e71554799 is the syntax

commit 752a8af7 (2022-07-03): fix #4315
closed 2022-07-03 by rusefillc