rusefi/rusefi #4322: Lua code causing boot loop

ElDominio, 2022-07-06:
This code

https://gist.github.com/ElDominio/8385f846d1e0bcc8d2e2569ff62d45e9

Causes rusefi (at least 4chan)  to be stuck at boot SOMETIMES.

Videos here: https://drive.google.com/drive/folders/1_tZcS8Qz7JpopSBae-BIwZ7rOK1sBMPq?usp=sharing

rusefillc, 2022-07-06:
@ElDominio what is a routh estimate of SOMTIMES? one stuck per 5 ign cyles or one stuck per 50 ign cycles?

@ElDominio do you have time to attempt reducing script complexity while it's still stuck sometimes? like for instance does it have same stuck ratio if you remove half of txCan? does it still get stuck if you remove getSensor?

mck1117, 2022-07-06:
On a boot where it isn't stuck, can you paste the results of these two commands to the ECU? `threadsinfo` and `luamemory`

ElDominio, 2022-07-06:
Will do that tomorrow then!

rusefillc, 2022-07-08:
@ElDominio please try latest firmware now that we have #4324 we hope to learn about the error

mck1117, 2022-07-08:
oh we tried on slack to no avail

ElDominio, 2022-09-03:
This is still broken, I have no idea what's going on. I'm going to swap out his ECU to see if that fixes it. Has only happened on Lancer

rusefillc, 2022-09-03:
@ElDominio does it happen on bench or only on vehicle?

@mck1117 would it help to ship to NJ or WA?

ElDominio, 2022-09-03:
I have no tested on bench, it only happens when he keys off, it apparently does not shut off correctly

Will have better answers on Monday

mck1117, 2022-09-03:
can you try and recreate a "lancer evo" on the bench with some relays configured the same as they are in the car?

ElDominio, 2022-09-05:
https://gist.github.com/ElDominio/8385f846d1e0bcc8d2e2569ff62d45e9/revisions

Added an if to check is getSensor(CLT) is returning nil, this stopped console flood when key off since it apparently returned a nil value when everything was off.

rusefillc, 2022-09-05:
Still open question what's the connection between Lua error and boot loop?

ElDominio, 2022-09-05:
From what I can tell:

When key off, LUA console flodded console with the message about getSensor(CLT) returned nil forever, which was interrupting main relay control, which then screwed up everything else

mck1117, 2022-09-05:
> which was interrupting main relay control

I don't think this is possible

ElDominio, 2022-09-05:
Well that's just me assuming, ever since I added the if statement to check for nil, it's been correctly shutting down (that was today, it was locking up before since)