rusefi/rusefi #4329: Miata Hellen (NB2) Alternate Accel Enrichment Table

duhafnusa4, 2022-07-07:
use MAP vs RPM instead of TPS vs TPS

requested from Gavin Miata (2001 VVT NB2 with lightweight flywheel aka fast revving engine)

duhafnusa4, 2022-07-07:
@BeerMoneyMotorsports

BeerMoneyMotorsports, 2022-07-07:
Do you have examples?  Other platforms using this? @mck1117 @blundar @OrchardPerformance will need to chime in

moistbrochure, 2022-07-07:
[image]

Here is an example of enrichment based on map from an old tune on https://msqur.com/view.php?msq=111 
Looks like mapdot is weird value, not actual map values.

moistbrochure, 2022-07-07:
The goal would be to do it in pressure vs rpm because it wouldn't rely on tps input an time. It would reply on change in rpm and map. However, I have no idea what I'm doing.

mck1117, 2022-07-07:
The problem with MAP is that it's slloooooow compared to TPS.

We already have wall wetting acceleration enrichment, which is the "physical model based equivalent" to what you're describing here. Can you post a current log/tune that you think shows a situation where there could be improvement?

OrchardPerformance, 2022-07-07:
Matt beat me to it, that is litterally done better by wall wetting and the reason for TPS based as well is to counter act the fact the air has to move and change pressure before the MAP sensor reads a change.