rusefi/rusefi #4333: Lua A/C Input

ElDominio, 2022-07-08:
The "setAcDisabled" parameter has no explanation on wiki. I think this requires there to be a physical A/C input, and then it cn disable with Lua.

There needs to be a replacement for A/C input thru CAN fully, since some cars send A/C request fully over CAN to the ECU, then the ECU decides if it activates physical A/C output. (Lancer X);
Also, there needs to be a setting to send an "A/C Output" over CAN, since the Genesis Coupe is like this (both request and output are done over CAN, going to the ECU since it knows all engine operating parameters)

rusefillc, 2022-07-08:
https://github.com/rusefi/rusefi/wiki/Lua-Scripting#setacdisabledvalue is not a function to reguest A/C - just added code documentation

so that's for vehicle without physical A/C button like BK2 couple roger that

Not https://github.com/rusefi/hellen154hyundai/issues/54 which is BK1

ElDominio, 2022-07-08:
BK2 has a physical button, but the A/C panel handles it and broadcasts an A/C request on CAN, which is what the ECU recieves. It also recieves an "a/c ok"  from the ECU and engages the compressor directly from the panel

BK1 is fully hardware based

rusefillc, 2022-07-08:
@ElDominio please test once snapshot builds

https://github.com/rusefi/rusefi/wiki/Lua-Scripting#setacrequeststatevalue

closed 2022-07-08 by rusefillc