rusefi/rusefi #4335: debug is set to idle mode, but none of the debug fields show anything

rusefillc, 2022-07-08:
https://rusefi.com/online/view.php?msq=1109

debug is set to idle mode, but none of the debug fields show anything.

https://rusefi.com/online/view.php?log=924)

rusefillc, 2022-07-08:
related to https://github.com/rusefi/rusefi/issues/3614

@stefanst please try new fancy "live data" in console

as for logs, please see relevant "view" dialog - all fields of all "view" are logged

there is a chance that usability of all NAMES should be improved but at least "view" diallogs in TS show those names

stefanst, 2022-07-08:
I believe this is the "live data"
[image]
It does not appear to contain any information about PID state.

rusefillc, 2022-07-08:
it sounds like @stefanst is not hunting for "anything" but specificaly PID state?

stefanst, 2022-07-08:
Correct. I was trying to tune idle PID. It is tricky without seeing what the PID is doing.

stefanst, 2022-07-08:
AS to improvements in console: I will not use console for things that should be done in TS. Only if it's not possible to make TS do what is needed will I consider using a second tool.

rusefillc, 2022-07-08:
"view" in TS is different from "live data" in console - maybe there is a confusion with terminology

"view" in TS shows only a list of variables with values

"live source code" in console shows actual source code with code flow branching

rusefillc, 2022-07-08:
in the mean time renaming "idle" debug to "stepper idle" since that's all debug does now

rusefillc, 2022-07-08:
@stefanst great news: your log file has PID info for a few of PID unconditionally regarless of debug mode

[image]

@stefanst please clarify if you are all set or what is still desired to be clarified or improved

closed 2022-07-08 by rusefillc