rusefi/rusefi #4336: idle clarify part 2

rusefillc, 2022-07-08:
new ticket to continue #4335 since amazing problem statement

Maybe we should get educating me out of the ticket realm. There are still a few things I don't quite understand and that make it harder for me to tune idle:
Is there a list explaining what the names of the parameters in the log/ TS gauges mean? I searched the Wiki using obvious seeming terms, but was unable to find anything
Is the idle-status error the difference between target rpm and actual rpm? If so, it seems to get calculated only occasionally.
The I-term in the log I posted (https://rusefi.com/online/view.php?log=924) varies between 0.0 and 0.3. It appears to be maxed out at 0.3, even though there is still an error. It has authority to change the value of the output between -5 and +10. What is the relation between I-term and output?
There appears to be no log field for the P-term
What is the difference between the open-loop idle slider and the offset value when in CL mode?

Also: One thing that tripped me up was a wrong AC input. The ECU though the AC was turned on and jacked up the idle accordingly. There was no way I could see in the log that this was happening, looking at the idle values. I was able to diagnose it, because I realized the AC fan was running constantly.
TLDR: In order to tune PID idle, I would like to see how my total output value is made up from it contributing elements.

rusefillc, 2022-07-08:
Yes, error is difference between target and actual. 

RE: A/C portion diag

your log has "targetRpmAcBump" which was since renamed to "idle: A/C bump", is that part of the ask @stefanst ? or does zero value of targetRpmAcBump in your logs not matching expectations?

your log also says "AC enabled"

stefanst, 2022-07-08:
Maybe I need to formulate better:
In order for me to troubleshoot idle with an idle valve, I would like to see the following in the log:
- current IAC setting
- all factors contributing to that setting in a manner that when I add them up a+b+c+.... = IAC
- current target idle rpm and contributing factors

Also an explanation how "Min", "Offset" and "Open Loop Idle Value" work. Especially the difference between the OL value and Offset seems not intuitive to me.  

I assume that Min/Max is that total PID output Min/Max, whereas we have separate Min/Max values for I-term only. But I am not certain. It could also be that Min/Max could be applied to each P and D separately. 

Tool tips could be more verbose. I particularly dislike the tips for "derivative filter loss" and "anti windup frequency"- they just state that 0.1 is a good value. 
The tip for "CicPID" could also be enhanced. What is it? At least say "Normally this should be FALSE" or TRUE, whatever the case may be. 

With what we currently have it is difficult for a normal tuner to get a car to idle well, because the parameters are not well explained and their effects can not easily be observed. You pretty much have to understand the code in order to be able to tune idle.

mck1117, 2022-07-08:
> Tool tips could be more verbose. I particularly dislike the tips for "derivative filter loss" and "anti windup frequency"- they just state that 0.1 is a good value.
> The tip for "CicPID" could also [...]

these crap need to go away - we've done plenty of tunes now with the fixed idle controller that don't use the weird cic pid mode

> Offset seems not intuitive

Offset needs to go away too - it was a shim before the current idle controller worked like it does

BeerMoneyMotorsports, 2022-07-09:
Id agree offset needs to go away. I see no point. Open loop slider is great and i have had great success with tuning idle with the current Setup.

BeerMoneyMotorsports, 2022-07-09:
[idle tuning.docx](https://github.com/rusefi/rusefi/files/9075678/idle.tuning.docx)
This is a combo of what dave wrote with some RE specific info i applied to it.  This will be a good start to a clear doc with the current setup.

mck1117, 2022-07-15:
>  The tip for "CicPID" could also be enhanced. What is it? At least say "Normally this should be FALSE" or TRUE, whatever the case may be.

Idle CIC needs to go away, as we've now seen good results without it in all the scenarios where it was thought to do something useful.

> Offset seems not intuitive to me

Will PR to force offset=0 and remove UI