rusefi/rusefi #4348: Launch control bug?

elmepro, 2022-07-15:
If launch control is set to 2000rpm, then in real world i got 4000rpm looks like its two times more rpm than set to TunerStudio.
Also can we get extra option on launch control (when launch control active add extra fuel in %)?

Car bmw m50b25 tigger wheel 60/2
Board: Proteus
FW:  20220702

If something needed let me known.

shadowm60, 2022-07-15:
Can you share tune?

rusefillc, 2022-07-16:
please share log of launch attempt as well

rusefillc, 2022-07-16:
@elmepro I have to request that one github issue coveres only well, one subject

elmepro, 2022-07-17:
Im sorry for late response a had 2 day drifting training. (absolutely love rusefi)

Tune is same as uploaded  here are log file
[launch control log.zip](https://github.com/rusefi/rusefi/files/9128263/launch.control.log.zip)

rusefillc, 2022-07-17:
@elmepro you say driftning on rusEFI and no photos/no videos? do you want to break my heart?

elmepro, 2022-07-20:
> @elmepro you say driftning on rusEFI and no photos/no videos? do you want to break my heart?

Ouh don't worry I'm all in to Rusefi need small touches to car wiring and i will add videos and review

rusefillc, 2022-07-20:
@elmepro it's a usability problem not a bug here

See https://rusefi.com/forum/viewtopic.php?p=45889#p45889

rusefillc, 2022-07-20:
will replace "range" with "adder" on the dialog labels

commit 3d2bb2a6 (2022-07-20): Launch control bug? "range" thing is confusing fix #4348
closed 2022-07-20 by rusefillc