rusefi/rusefi #4356: scaling of micro rusefi ADC gauges

mmatousek89, 2022-07-18:
when i remove rezistors and use AN8 as analog input, asign it to advanced-> analog inputs ADC1, scaling is wrong, i see only 2.9V when i aply 5v to AN8 pin, when i asign to sensors-> oil pressure reading is correct

rusefillc, 2022-07-18:
:( @mmatousek89 i am upset by what I persive as low quality of this ticket. Specifically I am hurt by the fact that it does not seem to mention "microRusEFI"

mmatousek89, 2022-07-18:
> :( @mmatousek89 i am upset by what I persive as low quality of this ticket. Specifically I am hurt by the fact that it does not seem to mention "microRusEFI"

i does mention "micro" in title ;)

mmatousek89, 2022-07-18:
> @mmatousek89 please clarify what is source of signal with MRE test? is it just floating wire measuring pulled-up voltage?

source of voltage is 10k potentiometer and pushbutton to gnd with 4k7 pullup to 5V

rusefillc, 2022-07-18:
@mmatousek89 i guess I am just overall tired and unhappy with HW drama

mmatousek89, 2022-07-18:
> @mmatousek89 i guess I am just overall tired and unhappy with HW drama

i dont know how HW can be related if in other part of software (setted up as oil pressure) input work correctly

mmatousek89, 2022-07-18:
when i asign
"advanced->aux input->ADC#1=AV5"
problem is still same,
gauge "sensors-raw->raw aux1 rawanaloginput1gauge" show 0-2,960V when potentiometer aply 0-5.000V to input AV5

mmatousek89, 2022-07-18:
in console i can see right value as "input" 

Aux analog ADC14 ADC_SLOW PC3 adc=2.96/input=4.97V/divider=1.67

so i assume that gauge is by mistake linked to "adc" instead of "input"

commit 90e98bac (2022-07-18): scaling of microRusEFI ADC gauges fix #4356
closed 2022-07-18 by rusefillc